paparazzi/paparazzi #1937: Potential support for 3DR SOLO?

podhrmic, 2016-11-03:
Do you think it makes sense to put effort into porting paparazzi to 3DR Solo drone? We have already support for pixhawk (and in fact Iris/Iris+ drones) that works well. Solo uses Pixhawk 2 (see http://www.proficnc.com/) which is potentially cool hardware, even though less appealing than Pixhawk 1 (more complex etc). Solo is on sale now (since the crash of 3DR) for ~$400 for the full kit. 

So I am curious - is anyone considering using the Solo for your project? Indeed Solo is discontinued but still potentially a good deal for the money.

gautierhattenberger, 2016-11-03:
It's not like I against a new board (especially if the only changes are the new board files), but the Solo doesn't seem a so nice frame: http://www.droneflyers.com/2015/08/3dr-solo-is-a-nogo-for-now/

podhrmic, 2016-11-03:
Hmm, that is a pretty sad reading. Also judging from the teardown videos of Solo it is also harder to customize (comparing to their previous Iris frame)

fvantienen, 2016-11-03:
But I think the potential for the Pixhawk 2 is really nice and this way you automatically also support platforms like the Solo.

kevindehecker, 2016-11-04:
I'm pretty sure we are not talking about the same Pixhawk2 here. The Solo with Pixhawk v2.0 has been "released" years ago through the 3DR Solo, the real upcoming Pixhawk (v2.1) is still to be released (early november they say, and not from 3DR).

http://store.mrobotics.io/product-p/3dr-cube-2.0-mr.htm

It seems at least the third IMU is not on the v2.0, not sure if there other differences.

podhrmic, 2016-11-04:
Hmm, on a side note - have you heard about this board? http://www.dronesmith.io/luci/
Not sure how it compares against Pixhawk cube (either version) but at least you don't need a breakbout board for it...

OpenUAS, 2016-12-20:
Indeed if bought in US one can get good deals on this 3DR airframe. Too many drones already in possession and to many board to convert, so no personal buy atm. But if you have one and willing to lend it out to us, the motivation would be there. 

At the moment the effort against the possible userbase is too small IMHO. 

Idea: We could open an issue of boards to convert to in order of deemed importance to PPRZ users with a checklist, like we did for Bebop features to support.

Feel free to continue the discussion on gitter.im also about the luci. I've closed the issue for now. Plz do not feel offended.

closed 2016-12-20 by OpenUAS