paparazzi/paparazzi #1957: Confusing bat_levels

EwoudSmeur, 2016-11-24:
Currently in use:
MIN_BAT_LEVEL
CATASTROPHIC_BAT_LEVEL
CRITIC_BAT_LEVEL
LOW_BAT_LEVEL
MAX_BAT_LEVEL

This is a bit confusing... especially since CATASTROPHIC_BAT_LEVEL and CRITIC_BAT_LEVEL do the same thing, but catastrophic is for fixedwing and the other for rotorcraft. I suggest getting rid of the catastrophic one and using the critic one for fixedwing as well, as it is nicely implemented in the electrical subsystem.

I am not really sure what min bat level is supposed to do...

fvantienen, 2016-11-24:
Maybe also supply defaults for 1-6S LiPo's? As these are most commonly used.

EwoudSmeur, 2016-11-24:
good idea!

flixr, 2016-11-26:
CRITIC_BAT_LEVEL and CATASTROPHIC_BAT_LEVEL are not the same, see https://wiki.paparazziuav.org/wiki/Airframe_Configuration#Battery

MIN_BAT_LEVEL is basically just there to "automatically" disable the bat checking in case you don't have vsupply connected on your board... this could be renamed to make it more clear.

EwoudSmeur, 2016-11-28:
It seems to me that they are more or less the same, consider:
https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/main.c#L326
vs
https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/fixedwing/main_ap.c#L650

flixr, 2016-11-30:
More or less, but not quite: CATASTROPHIC is more severe (will kill the aircraft) than CRITIC.
Also it is distinguished in the GCS, it warns on both levels: https://github.com/paparazzi/paparazzi/blob/master/sw/ground_segment/tmtc/server.ml#L558

EwoudSmeur, 2016-12-01:
Perhaps the behaviour is not exactly the same, but the general idea is: stop flying when reaching a certain battery voltage. So why have two different defines for that? And perhaps a more clear name would help, like FAILSAFE_BAT_LEVEL.

dewagter, 2016-12-01:
The killing (of the throttle) was very good for fixedwing, but in
rotorcraft that has a slightly different meaning...

-Christophe

OpenUAS, 2016-12-12:
Note: EXAMPLE 
on rotorcrat one can set flightplan behaviour if bat level CRITIC_BAT_LEVEL e.g. attitude only with fixed throtlle 30% (or whatever) Everyone is free to decide what critical meant to them oneself and at which level this occurs.

e.g. on CATASTROPHIC_BAT_LEVEL send out distress SMS with position from a telemetry device. 

Note, just examples to indicate those different levels have their real life use and place.

OpenUAS, 2016-12-12:
e.g. for one of the fixedwing min bat level we stop going up  and glide back home from uphigh.
Example: for one of the rotorcraft on MIN_BAT_LEVEL we let it drop freefall and if height > takeoffheight + 500m, deploy parachute, wait for attitude to be reasonable OK, release parachute, and fly on to land at base.

EwoudSmeur, 2016-12-12:
It seems to me those examples are not part of the normal code base... Of course anyone can define any behaviour when a certain bat level is reached, but this is about standard defines that, I think, any airframe has and should have. Now the part where I am concerned with is that it is not clear (it was not to me at least), what these defines do.

flixr, 2016-12-12:
So is it clear to you know and this is just a documentation issue, or would you like to change something?

EwoudSmeur, 2016-12-16:
I still think it is strange to go in to emergency mode for one firmware at catastrophic, and for the other at critic. But apparently there is not much enthusiasm to change this, so let's leave it be

closed 2016-12-16 by EwoudSmeur

flixr, 2016-12-16:
> I still think it is strange to go in to emergency mode for one firmware at catastrophic, and for the other at critic.

You treat this "emergency mode" as it were one and the same precisely defined thing. IMHO they are not equivalent:
- CATASTROPHIC_BAT_LEVEL
  - fixedwing simply kills the throttle
  - rotorcraft doesn't do anything special (killing throttle is not of much use here, you might as well just go on until the battery dies)
- CRITICAL_BAT_LEVEL
  - fixedwing has no built-in response here
  - rotorcraft optionally can go to FAILSAFE mode (slowly land) if you have FAILSAFE_ON_BAT_CRITICAL enabled

> But apparently there is not much enthusiasm to change this, so let's leave it be

This has nothing to do with enthusiasm, I'm honestly wondering what would be better...
I'm don't think that getting rid of the CATASTROPHIC level is the way to go here.
But we could e.g.
- provide a similar FAILSAFE_ON_BAT_CRITICAL on fixedwing
- add CATASTROPHIC to rotorcraft/electrical (to do what?)

flixr, 2016-12-16:
Improving the docs is always welcome :-)