paparazzi/paparazzi #1983: [Rotorcraft] Euler int integrator issues

kirkscheper, 2017-01-13:
As it is currently implemented, the integrator command will overflow when the integrated error is saturate and the integrator gain is above 512. The settings currently allow the user to use the slider to set a higher gain (which seems like not a good idea and caused us a headache). The settings should be updated so the user can't cause overflow in the integrator command. Additionally, the result of the product of the integrated error and gain is bound to only 10% of the maximum available command. It would be nice if this was more.

podhrmic, 2017-02-16:
@kirkscheper would limiting the I gain slider to 512 fix the problem? Or should we use 64bit int for the calculation to avoid the overflows altogether?

kirkscheper, 2017-02-17:
Not really. The point is that the max pprz command can't be achieved with the current CMD_SHIFT of 11. We are currently using CMD_SHIFT of 8 but changing this would require all control gains to be reduced by a factor of 8.

Changing the limit would fix one half of the problem.