paparazzi/paparazzi #1995: DSMX not working

kevindehecker, 2017-01-31:
Currently I'm not able to bind or get RC link externally binded DSMX (on a Devo 10). This is the case for me on a Pixracer as well as for @EwoudSmeur  with a Lisa mx, so it seems a software issue. (and @OpenUAS reported a similar problem)

DMS2 does still work.

OpenUAS, 2017-02-08:
Note that airframe must have <define name="USE_DSMX"/>  in the spektum module section.

podhrmic, 2017-02-09:
It would be good to add a note about this to wiki...

flixr, 2017-02-09:
Actually this should be documented more clearly in the spektrum module xml....
http://docs.paparazziuav.org/latest/module__radio_control_spektrum.html

kevindehecker, 2017-02-22:
We've just tried the USE_DSMX flag, but did not work. The RC is definitely 'binded', but still RC status in GCS remains NONE. Same steps with DSM2 give OK results.

podhrmic, 2017-08-26:
Can anyone reproduce these results?

OpenUAS, 2018-07-17:
Since we have all the TX, RX, and all autopilots and other needed hardware to retest, self assigned

OpenUAS, 2018-07-31:
Soon™...

gautierhattenberger, 2021-12-13:
timeout

closed 2021-12-13 by gautierhattenberger