paparazzi/paparazzi #2006: Paparazzi center doesn't compile the airfcraft after code change

podhrmic, 2017-02-11:
How to reproduce the issue:

- start paparazzi center and compile any AC (BUILD button)
- change something in the source code and save the chance
 click on BUILD again

This is the output:
```
#######################################
# BUILD AIRCRAFT=apogee_chibios, TARGET ap
#######################################
make[1]: Entering directory '/home/podhrmic/Documents/Paparazzi/paparazzi'
make[1]: Nothing to be done for 'radio_ac_h'.
make[1]: Leaving directory '/home/podhrmic/Documents/Paparazzi/paparazzi'
make[1]: Entering directory '/home/podhrmic/Documents/Paparazzi/paparazzi'
make[1]: Nothing to be done for 'flight_plan_ac_h'.
make[1]: Leaving directory '/home/podhrmic/Documents/Paparazzi/paparazzi'
make[1]: Entering directory '/home/podhrmic/Documents/Paparazzi/paparazzi'
GENERATE /home/podhrmic/Documents/Paparazzi/paparazzi/var/aircrafts/apogee_chibios/flight_plan.xml from
/home/podhrmic/Documents/Paparazzi/paparazzi/conf/: Error loading xml file /home/podhrmic/Documents/Paparazzi/paparazzi/conf/: Is a directory
Makefile.ac:216: recipe for target '/home/podhrmic/Documents/Paparazzi/paparazzi/var/aircrafts/apogee_chibios/flight_plan.xml' failed
make[1]: *** [/home/podhrmic/Documents/Paparazzi/paparazzi/var/aircrafts/apogee_chibios/flight_plan.xml] Error 1
make[1]: Leaving directory '/home/podhrmic/Documents/Paparazzi/paparazzi'
Info: module 'imu_common.xml' has been loaded several times for target ap, merging options
Info: module 'gps.xml' has been loaded several times for target ap, merging options
Info: module 'gps.xml' has been loaded several times for target sim, merging options
Makefile.ac:250: recipe for target 'ap.ac_h' failed
make: *** [ap.ac_h] Error 2
make: Leaving directory '/home/podhrmic/Documents/Paparazzi/paparazzi'
```

Something related to the conf file? @gautierhattenberger ?

flixr, 2017-02-11:
I can't reproduce this issue...

podhrmic, 2017-02-11:
try this:
- compile Quad Lisa M 2 (clean+build)
- go to Makefile.chibios (or Makefile.stm32 since that is what Quad_Lisa_M2 is using) and change anything in the makefile (can be a comment or so on)
- go back and hit BUILD
- you will see the mentioned error message

If I change the airfame config file or the board makefile, that doesn't seem to make a difference

gautierhattenberger, 2017-02-11:
Now flight plan is needed to build modules. For some reasons, it tries to rebuild it even if it is not necessary and fails as the flight plan name was not passed as a compile option. https://github.com/paparazzi/paparazzi/commit/498c326ff7e3cb29c82e9789dcde2e0d595174c0 should fix this for now.

flixr, 2017-02-12:
Ah, the issue is not when making code changes, but Makefile changes...
Merging the fix as it works here. (But yes... the dependency handling in the build systems could be improved).

closed 2017-02-12 by flixr