paparazzi/paparazzi #2012: Airframe file recursion

kirkscheper, 2017-02-15:
We often have a couple of different vehicles with slightly different settings but need to have a totally different airframe file with just the different settings. I know that it is possible to include a sub xml into an airframe file but it currently has limited application as modules and firmware are not parsed into the Makefile.

I think it would be very useful if not only the settings were loaded into the airframe.h but that the firmware and modules could be loaded as well. This would allow us to have true reuse of files and help us with maintaining some of the airframes.

@gautierhattenberger what do you think?

gautierhattenberger, 2017-02-15:
so you mean that xml files include in airframe files should be fully parsed, is that right ?

kirkscheper, 2017-02-15:
Indeed. If possible, if there are field defined in both files that the head
file define overwrites whatever was included.

dewagter, 2017-02-15:
The easiest solution was to run the gen_aircraft on both main and
include/calibration/parameter files. This was much easier than to change
the parser, and while you loose some ability to set the order of
modules/defines, for most this was no issue:
https://github.com/dewagter/paparazzi/commit/3beec0d6b69e674703b95c87bfc9da9bd7027b7a#diff-ce5e9b689f552a7fe22f11a5406e229fR167

-Christophe

gautierhattenberger, 2017-02-15:
Actually the airframe xml file is expended before generating the makefile (eg. building a firmware), so you can have firmware, modules and so on in an included file. But then the file that is passed to other programs (either from reading conf file or with a request to server) is not expended at the moment. Which means that it may fail (simulator, GCS,...).
I can probably fix that for most standard cases and tools.

gautierhattenberger, 2017-02-16:
See https://github.com/paparazzi/paparazzi/commit/869671dfc05bbd2ac9b5f55883c49d06a766d198 for a partial solution. Now you can compile simulate and use an airframe file only constituted of includes.
Tell me if this solves your problem (or a part of it) and I'll make a PR.

kirkscheper, 2017-02-17:
I tested your branch, indeed it looks like this implements what I meant. Unfortunately I am working on a branch a bit further away from master for a demo next week so can't directly test it for the real world use case I mentioned above.

I guess you can merge this PR and close the issue. If later I have more requests I will reopen the issue.

@gautierhattenberger thanks a lot.

gautierhattenberger, 2017-02-17:
closed with 5539bc81ace3a755553c78116f584153e2d1c2f4

closed 2017-02-17 by gautierhattenberger