paparazzi/paparazzi #2017: Incorrect baro frequency Apogee

EwoudSmeur, 2017-02-17:
For Apogee, the baro_board file is compiled before main_ap (fixedwing and rotorcraft), in which BARO_PERIODIC_FREQUENCY is defined.

This means that at this line (https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/boards/apogee/baro_board.c#L59), it is not yet defined, defaulting to PERIODIC_FREQUENCY for the MPL_PRESCALER. This means that the baro will be read far less often than is specified with BARO_BOARD_APOGEE_FREQ, for instance 512/50~10 times less then 5 Hz, so 0.5 Hz for the defaults.

Can/should main_ap be compiled before baro_board? Or do we just add the define to the baro_board file as well?

gautierhattenberger, 2017-02-17:
`BARO_PERIODIC_FREQUENCY` is also a compilation flag coming from https://github.com/paparazzi/paparazzi/blob/master/conf/firmwares/subsystems/shared/baro_board.makefile#L258 so this should overwrite the value of `main_ap.c` and `baro_board.c` as long as a baro board is used. Did you actually measure that the real reading frequency ? What does the `print config` say ?

EwoudSmeur, 2017-02-18:
I have both seen 0.5 Hz updates and the printed the prescaler define and saw it was 512/50. But that is the point, I am not defining the baro frequency, which is why it is not defined yet when baro_board is compiled. But later, it is defined by default in main_ap.c

flixr, 2017-02-19:
The compilation order has nothing to do with it, a define in a C file will never be available in another compilation unit (C file) anyway...
Seems though that the apogee/baro_board.c is the only files that uses BARO_PERIODIC_FREQUENCY outside of main.
So the quick fix would be to define the default 50Hz in the apogee/baro_board.c as well....

EwoudSmeur, 2017-02-19:
Ok, I did not know that. Then it should not be in the c file, but somewhere else, since it is used in multiple flies. Defining the same thing in multiple files seems like a source of error to me

flixr, 2017-02-19:
While maybe not the nicest solution it should be fixed with 37bf6e5

closed 2017-02-19 by flixr