paparazzi/paparazzi #2020: Bebop 2 firmware v4.0.3 breaks frontcam workings

OpenUAS, 2017-02-21:
The new Parrot release v4.0.3 firmware make sure the frontacm is not working anymore. The libhxxx.so is updated by Parrot. Issue already known for a while but to be able to track it a bit better added to this list and assigned to FvT. More details will be added to issue

OpenUAS, 2017-03-02:
libh1enc.so has a function change BTW compiler used by Parrot GCC: (Sourcery CodeBench Lite 2012.03-57) 4.6.3)

OpenUAS, 2017-06-08:
Discussed with @fvantienen, has access to 4.0.6 new .so and he is busy fixing it...

OpenUAS, 2017-06-08:
Status change since a lot of TU students depend on a fix. 
@fvantienen: any update on this?

wv-tud, 2017-06-09:
Not a permanent fix, but I have been collecting my bebop (2) updates related to vision in `wv-tud/bebop_vision_update`, it currently just uploads an older version of `libh1enc.so` from previous FW

gautierhattenberger, 2017-07-10:
@wv-tud @fvantienen is it still broken ? Do you have a proper fix ? If not to you want to integrate a workaround in the meantime ?

wv-tud, 2017-07-10:
I will be making a pull request for the current fix/workaround shortly.

gautierhattenberger, 2017-08-18:
The next stable release was planned these days, do you have something to fix this ? Thanks

OpenUAS, 2017-09-05:
there was a "dirty fix" in the wv-tud repo, now looking a newer firmware again v4.2.1 :(

dewagter, 2017-11-22:
This issue only should only affect hardware encoding which is not yet in master

OpenUAS, 2018-02-01:
@wv-tud  yes we know you are very busy, congrats for the swarm Demo. It would be great if you could dive into ther PPRZ code and help us fix or merge a few things, TIA

OpenUAS, 2018-12-04:
Some work done..

gautierhattenberger, 2021-12-13:
timeout, reopen if needed

closed 2021-12-13 by gautierhattenberger