paparazzi/paparazzi #2023: gps_datalink or natnet2ivy should detect when Optitrack substitutes/messes up Rigid Bodies

wv-tud, 2017-02-22:
When using multiple UAV/Rigid Bodies in Optitrack's the Motive software it sometimes switches/substitutes Rigid Bodies (e.g. UAV with ID 30 and UAV with ID 31 suddenly get each others position information). 

This should be checked either in natnet2ivy or in gps_datalink.

I currently made a workaround on my own branch checking if the position difference since the previous packet in gps_datalink.c is larger than a configurable threshold (e.g. 75cm), if so it will set the gps fix to NONE.

The results of Rigid body switching are quite severe - imagine 8 UAV mid-air when suddenly all the UAV are getting each others positions. To a certain degree this can be counteracted by either proper stickering of the UAV and settings in motive. However when increasing the number of rigid bodies the switching happens more and more frequent.

flixr, 2017-02-24:
This definitely needs to be fixed on the natnet2ivy side or ensuring that the tracking system can sufficiently distinguish the differently marked UAVs.

OpenUAS, 2017-03-02:
Will test Motive v1.10, according to optitrack multitracking AS's improved a lot from v1.9

OpenUAS, 2017-04-19:
Well nice new Prime 17W cameras are installed, just connected Motive v1.10. @wv-tud plz test current setup and  not share markers improvement on Motive side.

OpenUAS, 2017-05-19:
Just update to 1.10.3. Looks quite OK, but needs @wv-tud testing... soon™ ?

wv-tud, 2017-05-19:
@OpenUAS Will be testing on monday/tuesday with 6 if the zoo is available

OpenUAS, 2017-05-31:
There where no complaint anymore from anyone. Feel free to reopen issue, if it is still problematic, then We'll fix it together.

closed 2017-05-31 by OpenUAS