paparazzi/paparazzi #2037: IRIS AP build Error

winszenYoung, 2017-03-15:
I try to flash iris. I have a problem to build. 
It is complie error output

```
CC /home/kdy/dev/paparazzi/var/aircrafts/Iris/ap/subsystems/imu/imu_px4fmu_v2.4.o
subsystems/imu/imu_px4fmu_v2.4.c: In function 'imu_px4_init':
subsystems/imu/imu_px4fmu_v2.4.c:48:54: error: 'IMU_L3G_SPI_SLAVE_IDX' undeclared (first use in this function)
   l3gd20_spi_init(&imu_px4.l3g, &IMU_PX4FMU_SPI_DEV, IMU_L3G_SPI_SLAVE_IDX);
                                                      ^
subsystems/imu/imu_px4fmu_v2.4.c:48:54: note: each undeclared identifier is reported only once for each function it appears in
subsystems/imu/imu_px4fmu_v2.4.c:51:62: error: 'IMU_LSM_SPI_SLAVE_IDX' undeclared (first use in this function)
   lsm303dlhc_spi_init(&imu_px4.lsm_acc, &IMU_PX4FMU_SPI_DEV, IMU_LSM_SPI_SLAVE_IDX, LSM_TARGET_ACC);
                                                              ^
../../conf/Makefile.stm32:210: recipe for target '/home/kdy/dev/paparazzi/var/aircrafts/Iris/ap/subsystems/imu/imu_px4fmu_v2.4.o' failed
make[1]: *** [/home/kdy/dev/paparazzi/var/aircrafts/Iris/ap/subsystems/imu/imu_px4fmu_v2.4.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/kdy/dev/paparazzi/sw/airborne'
Makefile.ac:236: recipe for target 'ap.compile' failed
make: *** [ap.compile] Error 2
```

I follow this video
https://www.youtube.com/watch?v=L18yjer438w

How do I solve this problem?

winszenYoung, 2017-03-15:
this is output screenshot
[image]
[image]

gautierhattenberger, 2017-03-19:
You need to configure `IMU_L3G_SPI_SLAVE_IDX` in your airframe file. Not sure why there is no proper default value, I don't use the IRIS frame.
See: http://docs.paparazziuav.org/latest/module__imu_px4fmu_v2_4.html

flixr, 2017-03-19:
This is already fixed in latest master... you are still using v5.10 where you probably need to add the define as @gautierhattenberger said.

Closing as it should work fine with master.

closed 2017-03-19 by flixr