paparazzi/paparazzi #2040: Pixhawk fwb upload unreliable

podhrmic, 2017-03-19:
on Ubuntu 16.04 with the latest master, I am unable to flash the FBW through AP (using the `px4flash` module). It either receives ERROR from the AP, or freezes completely (have to restart the pprzcenter).

Tried on several Pixhawks, the same results.

When flashing the FWB with a JTAG (Black Magic probe), everything indeed works fine - so I am wondering if there is something with the `px4flash` module?

kevindehecker, 2017-03-21:
I'm having the same problem. As a work around; if you time it just right you can still upload to FBW. For me it is just after the second slow blink of the AP sys time led, but it depends how much you still need to build and how fast your laptop is.
If it didn't work make sure to kill the upload script process manually before attempting again (e.g. killall python ), because pressing stop from pprz center won't do it.

I'll try to look into this probably in two weeks or so.

podhrmic, 2017-03-21:
Cool, thanks @kevindehecker

podhrmic, 2017-04-14:
@kevindehecker any updates here?

kevindehecker, 2017-04-18:
I gave it a try, tried a few things,, but could not fix the problem. Then 2 weeks holiday :) Hopefully in the next few days I'll have time again.

OpenUAS, 2017-04-19:
Having Pixhawk, Pixracer and various PX clones, and not using JTAG on them we know annoyance: Timing the USB plug-in just in time etc.etc. As developers, well.. we manage; but it is for users quite an issue we experience every time. The first experience with PPRZ should be smooth(er). If one cannot upload without to much hassle, it is quite frustrating and users tend to look elsewhere for an open-source AP solution. So a little higher status label ;)

podhrmic, 2017-04-26:
Seems to be fixed with #2057 so closing.

closed 2017-04-26 by podhrmic