paparazzi/paparazzi #2041: GCS and RealTime plotter crash in Ubuntu 16.04

podhrmic, 2017-03-19:
Randomly and after a relative short time, GCS and Real Time Plotter silently crash in Ubuntu 16.04. Looks like OCaml or GTK+ is not a friend with the Ubuntu - can we resolve this somehow?

Having the GCS crash makes it hard to use for longer missions (operators get surprised when the GCS has to be restarter).

flixr, 2017-03-19:
I cannot reproduce this issue.... at least not is simulation, but that should make no difference as the data you get there is the same or actually even more...

podhrmic, 2017-03-20:
I ll investigate over the weekend and post more info.

podhrmic, 2017-03-20:
Here is what I have from a crashed plotter:
```
RUN '/home/podhrad/Documents/Paparazzi/paparazzi/sw/logalizer/plotter '

** (plotter:21284): WARNING **: (../../../lib/unity-gtk-action-group.c:355):unity_gtk_action_group_really_change_action_state: code should not be reached

** (plotter:21284): WARNING **: (../../../lib/unity-gtk-action-group.c:363):unity_gtk_action_group_really_change_action_state: code should not be reached
 Read Socket : Connection reset by peer

STOPPED '/home/podhrad/Documents/Paparazzi/paparazzi/sw/logalizer/plotter'
```

flixr, 2017-03-20:
Hm... have never seen that before... Do you have steps to reproduce this?

kevindehecker, 2017-03-21:
I have the same problem (at least with GCS). Disabling the battery read outs that the tudelft conf has by default made it much better (now crashes only once an hour or so, instead of after a few minutes tops). Not sure if related.

My steps to reproduce are basically: start GCS and wait for it.... Can reproduce in sim just as easily as with real drone, but the timing is completely random. (so the sim does not crash at the same time as with the real drone). Running the same logs etc on another computer does not crash it at all.

Does your laptop have a touchscreen by any chance? My laptop is, and our Outback field laptop is touched as well (literally and figuratively).

gautierhattenberger, 2017-03-21:
As a side note, I'm running Kubuntu 16.04 and I've never seen this. So the error is most likely related to unity as the error message suggests.

podhrmic, 2017-03-21:
I am running it on ThinkPad Carbon X1, but I ll try to use Gnome instead of Unity and see if it makes a difference.

flixr, 2017-03-21:
I'm also running unity and haven't been able to reproduce this with a simulation of Quad_LisaMX

OpenUAS, 2017-03-24:
Using Default Ubuntu 16.04.2 but also xubuntu  and mint on a touchscreen enabled devices. Running GCS for hours on end and with all kind of plotters tools etc. Never had an GCS crash since years. It is extremely annoying indeed. I run the GCSand all tools on at least a dozen different PC's even under OSX native, no issue. If you need testing help, make a not on this issue or gitter plz.

noether, 2017-03-24:
Xavier and me have the same problem. Ubuntu 16.10 and unity for me and Xavier with Kubuntu .

noether, 2017-05-08:
just fyi. It is still crashing in 17.04
any suggestion about how can I log the session, so in case it crashes again you can look at it?

gautierhattenberger, 2017-07-24:
Some updates here regarding GCS crashes: I have
- a laptop with Kubuntu 16.04 (KDE) that is working perfectly
- a laptop with Ubuntu 16.04 (Unity) that is crashing really fast each time

Running with debugger gives a segfault and this kind of information:
```
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
```
Also, it seems that it is always related to graphical element update from IVY callbacks. So there is a chance that Ivy ends up in a different thread than the Gtk canvas elements and that the thread protection is not properly done. But this doesn't explain why it was working before and why it has different behaviors between computers (and my colleagues have issues with KDE, so it is not related).

I tried to implement the `XInitThreads` call before running the GCS main loop. On the working laptop, it makes no difference (still working), on the other one, it is just crashing straight at startup...

As a conclusion: I'm even more confused than before...

podhrmic, 2018-04-14:
Update: this issue seems to be better ( at least I haven't seen any crash recently) with the recent Ubuntu updates. Proposing closing this issue as resolved?

kevindehecker, 2018-04-14:
Yeah, it seems to have solved itself :)

noether, 2018-04-14:
Yes, I agree. Magically I did not experience this issue for a looong time.

gautierhattenberger, 2018-04-14:
cross fingers for 18.04LTS !

closed 2018-04-14 by gautierhattenberger