paparazzi/paparazzi #2058: BMP V2: need new flash mode: SWD3PIN

dewagter, 2017-04-28:
Black Magic Pro V2 drives the SWD pins with the voltage it detects from the board by default.

When using 3-pin SWD (e.g. lisa-s, lisa-mx-s, stereoboard, ...) the target voltage is not sensed and on new JTAG this means the SWD pins are driven at zero volt.                   

Solution is simple: **mon tpwr enable**

But on 10 pin jtag boards, this is not good, as this will short the BMP 3.3V and Lisa-M 3.3V. This also means the Lisa-M will not completely reset when the JTAG cable is connected while the lisa is unpowered (power comes from JTAG board).

Solution: new flash mode:

SWD10PIN  (do not enable power)
SWD3PIN (do enable power)

podhrmic, 2017-08-26:
Is this fixed with 6c484d494f9d3b8b1174fc8b104788ccd4c398cb ?

kirkscheper, 2017-10-27:
closed with #2104

closed 2017-10-27 by kirkscheper