paparazzi/paparazzi #2064: ahrs_int_cmpl_quat jumps on bebop1

EwoudSmeur, 2017-05-14:
I am flying with a Bebop1 version 4.0.5 (current limit is 4.0.4) with ahrs_int_cmpl_quat and magnetometer turned of (RC flight). I am logging the attitude at full speed (512 Hz) and I have seen jumps in the attitude (roll pitch and yaw, and all quaternion entries) from one sample to the next. This has happened in multiple flights after +- 60 seconds of flight. The state body rates do not show such a jump. When landed, I can see the attitude gradually go back.

Not sure what could cause this... I am running slightly modified control code.

flixr, 2017-05-14:
Does this also happen with older bebop versions?
You are not using any heading update (since you disabled mag)?
The aligner is ready when you start flying?
Did you also log the IMU?

EwoudSmeur, 2017-05-14:
I can test more after tomorrow, but I am not sure I can test with an older version.
Indeed I just wanted to fly RC inside, without magneto. I don't think it is related to this, but can investigate.
Indeed, I always wait for the aligner.
I did not log the IMU directly, but I did log the bodyrates. Would be interesting to see what the accelerometer is doing, or if there are timing problems.

I can do tests on tuesday, but I thought I'd just report this

EwoudSmeur, 2017-05-16:
I had the bebop on the desk just now, same settings, logging IMU. I don't see anything happening to gyro, accel or mag when I see these sudden attitude changes. I copied a picture here, hope it will show:
[image]

EwoudSmeur, 2017-05-16:
I think I may be spamming you guys, but I found this:
[image]
in gyro_cb function the value of dt is 0...  

So I suppose something must be going wrong in imu_bebop.c, right? What could cause the callback to be called with the same time? And why does he sometimes seem to 'miss' an event and then catch up?

fvantienen, 2017-05-16:
The resolution of the `sys_time` is `SYS_TIME_FREQUENCY`. So prolly it is not called at the exact same time, but faster than the measurable frequency. Next to that it could also be caused by the fact that we don't have a limit on the `event` thread, and the `sys_time` thread could be lacking behind.
You should check the frequency settings of the MPU and also the `SYS_TIME_FREQUENCY`.

EwoudSmeur, 2017-05-16:
Thanks Freek, this indeed seems to be the case:
#define SYS_TIME_FREQUENCY (2 * PERIODIC_FREQUENCY)
Is anything limiting us to ramp this up with, say, a factor 1000 or something like this?

gautierhattenberger, 2017-05-16:
I don't know for Linux based OS, but on MCU like STM32F4 it have an impact on some sys_time functions. But 10 times is probably fine (ChibiOS is set to 10kHz for sys time freq).
Also, you can set a fixed value for the IMU dt. It will avoid the jumps, and is finally more accurate than the automatic dt detection.

EwoudSmeur, 2017-05-17:
So let me get this straight... by default we are calling a function called get_sys_time_usec that is actually only counting half milliseconds (which is not really documented). We are then proceeding to use this to calculate one of the most important time differences in the ap. And then it actually says above this function that it will overflow in 70 minutes.
I can see some problems here :smile: 
In fact, this time function is used all over, would it not be better to make a function that calculates a dt with overflow detection? Or should it just be uint64?
And what exactly would be the problem if we go above 10khz sys time freq?

flixr, 2017-05-17:
No that is not correct, get_sys_time_usec really has a usec resolution (at least on the bare metal MCUs it actually uses the cpu ticks and converts that to usec, see [get_systime_usec](http://docs.paparazziuav.org/latest/stm32_2mcu__periph_2sys__time__arch_8h.html#af3a1d6edcddd66ef6e1ac49dc3e99abf)

And for computing the dt the overflow of this usec return value doesn't matter, as the difference is still correct.

I think there is some confusion about SYS_TIME_FREQUENCY here:
- it just means that our virtual timers (the ones created with `sys_time_register_timer`) have a resolution of 1 / SYS_TIME_FREQUENCY

In the linux implementation of [get_systime_usec](https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/arch/linux/mcu_periph/sys_time_arch.h#L39) it doesn't use the cpu ticks but only the `sys_time` struct which is updated in a separate thread at SYS_TIME_FREQUENCY.

So it seems to me that the most important thing that really needs to be improved here is the linux implementation of `get_systime_*`, a higher SYS_TIME_FREQUENCY would just make this problem less visible instead of solving it.

EwoudSmeur, 2017-05-17:
Ah yes good point about the overflow. So... it is not completely clear to me where these cpu ticks are updated, and if we can use this for linux as well. Do you have a view of what specifically should be changed?

If for the linux part we first calculate ticks from `clock_gettime(CLOCK_MONOTONIC, &now);`, and then convert it back to a a time, why don't we just call this function in `get_sys_time_usec`?

EwoudSmeur, 2017-05-17:
Ok thanks for the help, I'll try to do this soon.

EwoudSmeur, 2017-05-24:
https://github.com/paparazzi/paparazzi/pull/2067
This seems to fix most of the problem, still some funky jumps in the timing, but not zero any more:
[image]

OpenUAS, 2017-05-31:
Can re-fly RC(&Auto) firmwares BB1 v3.3 & v4.0.5 & BB2 v3.9 & v4.0.5 this week if needed.

EwoudSmeur, 2017-06-23:
I still have a problem... I have had three times that my bebop suddenly had a jump in the state estimate and crashed as a result. The third time I was logging the dt that is being used in the gyro callback function:

[image]
(x axis is time in seconds calculated from the log sample number, y is system time difference in seconds)

I was also logging the system time, and here I see the same 0.25 second jump, so both the logger as well as the gyro function were not called for 0.25 seconds. My hypothesis is that the whole autopilot froze... Needless to say this is suboptimal for the state estimation. Any ideas what might be causing this, or how/what to investigate further?

EwoudSmeur, 2017-06-26:
Now I am thinking this might be drone specific, maybe a process that is running in the background. I did a factory reset and now this particular problem seems to be gone.

podhrmic, 2017-06-28:
Can we close this issue then @EwoudSmeur ?

closed 2017-06-30 by EwoudSmeur