paparazzi/paparazzi #2068: RTP streaming problem on ARDrone2

tomvand, 2017-05-30:
Using the master branch of Paparazzi, RTP video streaming does not seem to work on the ARDrone2.
I am using the examples/ardrone2.xml airframe which contains the video_thread module and video_rtp_stream. When I open the .sdp file with VLC to view this stream, I only receive a single frame before the video stops. VLC's media statistics show that only one video block was decoded. I added a printf to rtp_packet_send(...) and verified that the drone *is* sending new packets.

This problem did not occur with v5.10 of Paparazzi. Overwriting modules/computer_vision/viewvideo.c, /lib/encoding/rtp.c and .h with their v5.10 versions also fixes the problem.

I'm using Ubuntu 16.04 with VLC 2.2.2.

OpenUAS, 2017-05-31:
Yes, now looking into it...

ETitus, 2017-06-06:
I've had the same problem as well at some point when I pulled master. Before that I was on v5.10. 

I assumed it was an error at my end though, as **closing VLC and reopening the .sdp file actually shows you a new single frame**. However, now it seems a true error.

OpenUAS, 2017-06-15:
@ETitus Thanks, yes it is a real issue on ARDrone2, will test on BB1 and BB2 asap. Assigne issue to self for the time being of retesting. It would be great if @wilcoschoneveld could look into it.

dewagter, 2017-06-16:
RTP is super dependent on time. Real-time Transport, in fact, has
everything to do with time. The default behavior for a viewer is to buffer
until it is sure it can show the stream without hesitations. So for a few
seconds, the maximal download delay is analyzed and the stream is shown
with this delay (so the buffer never depletes and the viewer can show the
steam without interruption).

see:
https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/computer_vision/lib/encoding/rtp.c#L133

Most likely something with frame time here...

-Christophe

dewagter, 2017-06-16:
> master/sw/airborne/modules/computer_vision/lib/encoding/rtp.c#L133

Actually: I think I got it:

https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/computer_vision/lib/encoding/rtp.c#L110

the seconds must be multiplied by 1000000 to get microseconds. otherwise
you never reach 1 second.

Titus, Tom and/or Erik, can you try this?

Previously it had the time move slower than real-time so the video was
always outdated and that caused most viewers to show it with less delay.
Maybe this can still be an option with a define for instance:

https://github.com/paparazzi/paparazzi/commit/d64675e4865d01ae54965de97b19b82151ee2c55#diff-5d33e46aed3ab4703018c1a4680eae37L111

-Christophe

OpenUAS, 2017-08-16:
@dewagter Missed your comment... Will test

lethargi, 2017-09-06:
I tried a number of things based on the comment and a small talk with @dewagter. I restored the old expression for frame time in the call to *rtp_frame_send* in the *viewvideo.c* file.

dewagter, 2017-09-06:
Did it work?

lethargi, 2017-09-06:
No. It didn't.

dewagter, 2017-09-07:
can you log the time stamps returned by get-time and the uin32_t that you
now create? (e.g. printf and telnet)

-Christophe

lethargi, 2017-09-07:
Frame time with *(img->ts.tv_sec * 1000000 + img->ts.tv_usec)*
```
Frame time:482411469
Frame time:483312428
Frame time:484213384
Frame time:485114342
Frame time:486015301
Frame time:486916256
Frame time:487817215
Frame time:488718174
Frame time:489619132
Frame time:490520088
Frame time:491421047
Frame time:492322002
Frame time:493222961
Frame time:494123917
Frame time:495024875
```

Frame time with *((img->ts.tv_sec  % (256 * 256)) * 900000 + img->ts.tv_usec* 9/100)* (the suggested correction)
```
Frame time:75624430
Frame time:76525389
Frame time:77426347
Frame time:78327303
Frame time:79228262
Frame time:80129220
Frame time:81030179
Frame time:81931137
Frame time:82832096
```

dewagter, 2017-09-07:
1000000 should be 100000

Hundred thousand, not a million.

lethargi, 2017-09-07:
I tried with *(img->ts.tv_sec * 100000 + img->ts.tv_usec)*. It did not work. I got the following frame time output with your fix

```
Frame time:888178250
Frame time:889079206
Frame time:889980159
Frame time:890881115
Frame time:891782068
```

Not sure if it is relevant, but my sdp file has the following text
```
v=0
m=video 5000 RTP/AVP 26
c=IN IP4 0.0.0.0
```

dewagter, 2017-09-07:
This counter corresponds to 1 frame every 9 seconds...

dewagter, 2017-09-07:
Microseconds should be still devided by 10.

Counter should be hundredthousands of a second. So sec*100 000+microsec/10.

At 25Hz video the counter should increase about 4000 counts every frame.

dewagter, 2017-09-07:
https://tools.ietf.org/html/rfc3550#section-5.1

 timestamp: 32 bits
      The timestamp reflects the sampling instant of the first octet in
      the RTP data packet.  The sampling instant MUST be derived from a
      clock that increments monotonically and linearly in time to allow
      synchronization and jitter calculations (see Section 6.4.1).  The
      resolution of the clock MUST be sufficient for the desired
      synchronization accuracy and for measuring packet arrival jitter
      (one tick per video frame is typically not sufficient).  The clock
      frequency is dependent on the format of data carried as payload
      and is specified statically in the profile or payload format
      specification that defines the format, or MAY be specified
      dynamically for payload formats defined through non-RTP means.  If
      RTP packets are generated periodically, the nominal sampling
      instant as determined from the sampling clock is to be used, not a
      reading of the system clock.

dewagter, 2017-09-07:
https://en.wikipedia.org/wiki/RTP_audio_video_profile

protocol 26 (Jpeg) has a clock rate of 90 000

conclusion:

 - clock that increments monotonically and linearly in time
 - not a reading of the system clock
 - counting using a 90 000 samples per second counter

dewagter, 2017-09-07:
@lethargi could you try my pull request #2110

lethargi, 2017-09-07:
Thanks Christophe for all the guidance!

I was trying the solution proposed by @tomvand. He said that using the files from v5.10 work. I copy pasted the files from v5.10 and smoothed out the edges and it works. Although there is a few seconds of delay. My stupid approach now will be to diff all the relevant files (*viewvideo* and *rtp*) and try to reverse engineer the cause of the problem.

I do not know exactly how to get a pull request on my local repo, so I manually made the fixes you proposed, and there seems to be no video output at all; not even the one frame that comes out without bugs. I cannot be fully sure if I made the corrections properly

OpenUAS, 2017-09-08:
@lethargi "...Although there is a few seconds of delay..." using vlc as client ? try mplayer
BTW @lethargi you could checkout the fix_rtp_stream branch and try that something like this from top of my head:  ($ git remote update && git fetch --all && git checkout fix_rtp_stream && make && make clean)

lethargi, 2017-09-08:
@OpenUAS I tried mplayer but it has more lag than vlc. And thanks for telling me how i can get the patch, but what i really wanted to do is only checkout the fixed files and merge those into the branch where i am developing. I just used `git checkout source_branch <paths/to/file>`, which i got from [here].(http://jasonrudolph.com/blog/2009/02/25/git-tip-how-to-merge-specific-files-from-another-branch/)

@dewagter just verified that your patch works for AR Drone 2. Hartelijke bedankt!

OpenUAS, 2017-09-08:
Working now, with ARdrone2 will now test Bebop and Bebop2... then pull req