paparazzi/paparazzi #2072: Switch statement unreachable

podhrmic, 2017-06-08:
@gautierhattenberger this is what came from coverity scan:

[image]

Could you have a look at it and see if it s a false positive or not? I wasn't sure.

gautierhattenberger, 2017-06-13:
It probably depends if `USE_STABILIZATION_RATE` is defined or not (used when setting horizontal mode), but I'm not even sure it is the problem. Unless coverity can provide more information, I don't know how to fix this. Doing rate mode with z_hold is probably a bad idea from the start anyway...

podhrmic, 2017-06-13:
In that case Coverity probably complained that the mode cannot be reached without `USE_STABILIZATION_RATE` and I would assume that neither of the test airframes have this flag. 

I ll mark it as false positive, but on the other hand - shouldn't we remove the `RATE_Z_HOLD` mode in that case (being a bad idea)?

podhrmic, 2017-06-13:
Closing as false positive.

closed 2017-06-13 by podhrmic