paparazzi/paparazzi #2073: actuators i2c - missing break

podhrmic, 2017-06-08:
@flixr do you think this is intentional? It is not written in the most clear style, so I am unsure without actually testing it:

[image]

EwoudSmeur, 2017-06-13:
hm I agree it is not the most clear code, sorry about that...
For this specific break, I don't think it is needed as after executing case 4 he should stop anyway, whether it is this cycle or the next. I remember testing it and if I remember correctly it worked.

gautierhattenberger, 2017-06-13:
I also don't think it is strictly necessary to have a `break` to make this work. It should probably be rewritten, on the over hand it is working and unless someone has time to clean the code and test it, I suggest to keep it this way.

podhrmic, 2017-06-13:
OK, thanks for the feedback. I added a note there to keep a record for the future and to keep Coverity happy: https://github.com/paparazzi/paparazzi/commit/124f04fb0f7bc9f8d19f520cf4744b9b7a3bcf82

closed 2017-06-13 by podhrmic