paparazzi/paparazzi #2080: Support for loading and running deep learning models

wilcoschoneveld, 2017-06-21:
I'm working on another implementation of optical flow with the use of a convolutional neural network (CNN). I have a prototype of the network working and want to implement this in paparazzi. Unfortunately this is not straight forward and it would be great if I could get some input on how to continue.

Some details about my prototype architecture; It is currently a simple sequential network with 3-4 convolutional layers followed by a fully connected layer. Input is two 64x64 images, output is an optical flow vector (2 floats). I have created and trained the prototype with TensorFlow 1.2.0, but i'm not restricted to use any other framework.

**I'm looking to implement some module in paparazzi which allows evaluation of a frozen deep learning model.**

OpenCV 3 has a [deep neural network module](https://github.com/opencv/opencv_contrib/tree/master/modules/dnn) which is in active development. I've [tried](https://github.com/opencv/opencv_contrib/issues/1241) to import a freeze of my graph but the tensorflow importer seems to be a bit outdated. Also, I don't think it's the best idea to work with OpenCV for this since there are more areas in paparazzi (besides vision) which machine learning could be applied to.

[Tiny-DNN](https://github.com/tiny-dnn/tiny-dnn) is a header-only learning framework which supports both execution and training. There are no required dependencies but recent development is done with C++14 features (can we use this?). Development of their tensorflow loader has [halted](https://github.com/tiny-dnn/tiny-dnn/issues/439) but they are working on 'some interfaces to import models from any type of framework'. They have a Caffe importer which I'm going to try next, but I have to start using Caffe itself first.

Any inputs on this?

rmeertens, 2017-06-21:
Tensorflow itself can use C++ to load pretrained graphs. You can compile it with Bazel... Compiled C++ stuff can be added to Paperazzi, although I'm not an expert in this. For an example you could check how we added OpenCV. 
Does this Stackoverflow answer help? https://stackoverflow.com/questions/33620794/how-to-build-and-use-google-tensorflow-c-api

wilcoschoneveld, 2017-07-09:
For the past two weeks I've been trying to compile tensorflow for the bebop. There is a bazel target `//tensorflow:libtensorflow.so` which allows you to build a shared library with a C api. This seemed to be an ideal candidate. There is also a `libtensorflow_cc.so` target to build a C++ lib. Setting up bazel to cross-compile is something I never attempted before and I didn't get it to work. I followed a cross-compiling approach as implemented in https://github.com/snipsco/tensorflow-build, and setting it to the `parrot-tools-linuxgnutools-2012.03_1.0.0-1_i386` toolchain, but I think the toolchain is too outdated to built it properly. For now I'm giving up on this.

gautierhattenberger, 2017-07-10:
Should we close this for now, then ?

closed 2017-07-11 by wilcoschoneveld