paparazzi/paparazzi #2086: xbee_api telemetry not working on Umarim

podhrmic, 2017-07-06:
To reproduce (on latest master): 

1. `./start.py ` and choose `conf_examples`
2. select AC `Umarim_Lite`
3. change telemetry to `xbee_api` (here https://github.com/paparazzi/paparazzi/blob/master/conf/airframes/examples/umarim_lite_v2.xml#L21 )
4. compile and upload

The Umarim won't blink or respond in any way. If you do the same with `transparent` telemetry, it, works perfectly fine.

Ideas what can be wrong? @gautierhattenberger

podhrmic, 2017-07-06:
xbee_api compiles and works on Lisa M, so it seems to be related to the LPC arch only.

gautierhattenberger, 2017-07-07:
I have just tested on a plane with Umarim xbee-api and latest master. Everything is working correctly.

podhrmic, 2017-07-07:
Thanks for testing @gautierhattenberger ! It must be some hardware problem then.

closed 2017-07-07 by podhrmic