paparazzi/paparazzi #2090: Bounds error in NPS accelerometer simulation for ARDrone2

tomvand, 2017-07-21:
I have found a possible bug in the nps accelerometer model or the nps sensor parameters for the ARDrone2:

In [`nps_sensor_accel.c`](https://github.com/paparazzi/paparazzi/blob/master/sw/simulator/nps/nps_sensor_accel.c#L39), the accelerometer values are *first* offset by their neutral value `NPS_ACCEL_NEUTRAL_X/Y/Z` (equal to `IMU_ACCEL_X/Y/Z_NEUTRAL` = 2048 as set in the airframe file) in line 39 before being saturated in line 56.

In [`nps_sensors_params_ardrone2.h`](https://github.com/paparazzi/paparazzi/blob/master/conf/simulator/nps/nps_sensors_params_ardrone2.h#L38), however, the bounds `NPS_ACCEL_MIN/MAX` do not seem to take this offset into account. In fact, the upper bound `NPS_ACCEL_MAX` (=2047) is smaller than the neutral value `NPS/IMU_ACCEL_X/Y/Z_NEUTRAL` (=2048). This causes the the accelerometer to saturate, especially when it already has an offset from `BODY_TO_IMU`.

As far as I see it, the bounds in [`nps_sensors_params_ardrone2.h`](https://github.com/paparazzi/paparazzi/blob/master/conf/simulator/nps/nps_sensors_params_ardrone2.h#L38) should be set to
```
#define NPS_ACCEL_MIN 0;
#define NPS_ACCEL_MAX 4095;
```
Can anyone confirm this?

OpenUAS, 2017-09-08:
Assigned to self to see to validate fix proposed. Thx @tomvand for investigating

kirkscheper, 2017-10-15:
Should be addressed with #2120

closed 2017-10-18 by kirkscheper

tomvand, 2017-10-19:
#2120 has not fixed the problem (it also did not change nps_sensors_params_ardrone2.h), checked by plotting the scaled sensor values:
[image]

kirkscheper, 2017-10-19:
Indeed, I had it in my branch but forgot to add it to the PR. tnx for the review.

commit 5a985ef6 (2017-10-19): Fix #2090  (#2122)

* Add drag to gazebo ardrone model

Simple linear drag with guessed coefficient, not super accurate
but better than no drag at all...

* Fix #2090

Set accelerometer sensor bounds to correct values.