paparazzi/paparazzi #2092: No erase SD card option in sdlogger_spi_direct

matejkarasek, 2017-07-25:
When using sdlogger_spi_direct.c, there is currently no option to erase the SD card (unhandy when the SD card is soldered to the autopilot...). In v5.8, the SD card would erase when a new log was started after an autopilot restart. The observation made on a Lisa/S and reproduced by 'woeileong1104'.

OpenUAS, 2017-09-05:
True, have the hardware now to test and find fix.

OpenUAS, 2017-09-05:
Assigned to self since almost no one else on the planet has this hardware combo. No milestone to target ;)

matejkarasek, 2017-09-05:
Thanks, would be very helpful!

If you are already looking into it, there is another bug. Only the first 42 logs work correctly, starting from log 43 something overflows and size displayed is incorrect... Erasing the SD card solves it, but that is currently only possible by going back to v5.8.

[image]

OpenUAS, 2017-10-31:
FYI LisaM AP with SP SD still on desk in and todo list...

OpenUAS, 2018-07-17:
still...just a lot of tasks...

OpenUAS, 2018-12-04:
Will likely be looked into by BS in Jan2019