paparazzi/paparazzi #2097: Optical flow code sends NaN to INS, causing it to reset

ETitus, 2017-08-03:
https://github.com/paparazzi/paparazzi/blame/master/sw/airborne/modules/computer_vision/opticflow/opticflow_calculator.c#L376
Recently added in #2079 , this line of code fills the result structure with 0's, including the FPS parameter when not enough corners are found. This happens when the drone is close to the ground, or no features are visible.

This FPS parameter is then used in the Kalman filter:
https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/computer_vision/opticflow/opticflow_calculator.c#L802
Where it tries to compute 1.0f / fps

Naturally this is not intended behavior, as it results in NaN velocities that the kalman filter outputs. These are send over the ABI message into the INS, causing the INS to go: "I don't know what to do, I'll just make everything 0". 

The result is the INS flipping between 0 and the actual values received from OptiTrack for example.
The behavior can be seen in this video:
https://www.youtube.com/watch?v=8EhzH8DJy_w

When flying this gives some trouble, i.e. twitching, but the real trouble comes when landing, at which point the drone really flips out and flies away out of control.

This is confirmed for both Ardrone2 and Bebop, that I tested together with @OpenUAS over the last two days.

@manoskyr I am not entirely sure which variables you want to reset and why. Can you explain this perhaps? We might be better off just resetting certain variables or not even resetting them. Alternatively it might be possible to skip the kalman step too when not enough corners are found?

ETitus, 2017-08-04:
With https://github.com/paparazzi/paparazzi/blame/master/sw/airborne/modules/computer_vision/opticflow/opticflow_calculator.c#L376 uncommented I decided to test https://github.com/paparazzi/paparazzi/pull/2087 and I found more NaN values seeping through from the optical flow code.

I am not entirely sure why this is happening, but my (partial) analysis:
It originates from this division:
https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/computer_vision/opticflow/size_divergence.c#L117
Somehow after the vision updates it is possible for distance_1 to be 0.

It is of course possible to make a check if distance_1 is 0, and not divide by it if that is the case, but I am not sure what the value of divs[sample] should be set to in that case, 0? This "solves" the NaN, but the computed result might be wrong. 
Optionally a new pair of i,j can be forced by returning? 

However, as this never occurred in previous versions of the optical flow code, I dived deeper:

distance_1 = 0 means that 
```
dx = (float)vectors[i].pos.x - (float)vectors[j].pos.x;
dy = (float)vectors[i].pos.y - (float)vectors[j].pos.y;

```
there is a vector pair i,j for which the positions in both x and y direction are the same. Seemingly unrelated I actually also looked at distance_2, which is also 0 in the case that distance_1 is zero.
This leads me to believe that duplicate vectors are present from https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/computer_vision/opticflow/opticflow_calculator.c#L387

However, I am not sure if that is the case, and why that would be. Any ideas? 

Confirmed on Ardrone2 and Bebop.

manoskyr, 2017-08-07:
Sorry about the late response, been on holidays for the last week. Filling the result structure with zeroes was supposed to avoid sending the same velocities (the ones of a previous result) over and over while there are no corners detected. The idea was to have no result when no corners would be detected. I probably missed the fps division in the kalman filter so that makes it a problem. Manually setting only the velocities to zero (so no memset) would probably be a non elegant but working solution.
Unfortunately I don't have access to drones in my current location to resume testing so I can only keep working on a theoretical basis. Please keep me posted on any update.

kirkscheper, 2017-10-15:
This should be fixed with #2118

closed 2017-10-18 by kirkscheper