paparazzi/paparazzi #2112: [autopilot] Rotorcraft firmware in_flight check assumes you have an accurate speed estimate

kirkscheper, 2017-09-12:
The in_flight check currently checks that the thrust setting, speed and acceleration are below certain limits. This however assumes that the vehicle has an accurate speed estimation, which is not always true. I recommend changing this to check only for thrust and accel and/or do something with the motors_on boolean.

EwoudSmeur, 2017-09-14:
In what case would you have a bad speed estimation?

kirkscheper, 2017-09-14:
Any passively stable, non quad, non GPS or optical flow vehicle... So
Delfly ;p

dewagter, 2017-09-14:
Be a bit careful when pushing changes to master as this has crashed many
quads so far when wrong detections were made inflight.

-Christophe

EwoudSmeur, 2017-09-14:
Doesn't the Delfly have a baro?

gautierhattenberger, 2017-09-15:
The thresholds are configurable, so you can set something high enough in your airframe to disable some of the checks

kirkscheper, 2017-10-15:
Yea of course, dumb of me.

closed 2017-10-15 by kirkscheper