paparazzi/paparazzi #2116: sending arrays by ABI

knmcguire, 2017-10-11:
Hi all,

For my latest pull request (#2114), I would like to send arrays by ABI. However, when I try to compile the AP/NPS, it gives me a whole list of errors (see below). I was expecting that the same capabilities would be possible for ABI as for pprzlink, however paparazzi seems to not correctly compile it. For pprzlink for instance, the abi message would not only require the array, but also the size of it. The compiled sendABI message does not seem to do that.

Anybody who can take a look at it? Thanks!

__________________
_Error message during compiling the AP/NPS_

In file included from /home/knmcguire/paparazzi/sw/airborne/subsystems/abi.h:32:0,
                 from /home/knmcguire/paparazzi/var/aircrafts/ardrone2_rangesensors_gazebo/nps/generated/flight_plan.h:10,
                 from /home/knmcguire/paparazzi/var/aircrafts/ardrone2_rangesensors_gazebo/nps/generated/settings.h:13,
                 from subsystems/settings.c:29:
/home/knmcguire/paparazzi/var/include/abi_messages.h:60:72: error: expected ';', ',' or ')' before 'distance'
 typedef void (*abi_callbackRANGE_SENSORS)(uint8_t sender_id, int16_t[] distance);
                                                                        ^
CC /home/knmcguire/paparazzi/var/aircrafts/ardrone2_rangesensors_gazebo/nps/subsystems/commands.o
/home/knmcguire/paparazzi/var/include/abi_messages.h:371:79: error: unknown type name 'abi_callbackRANGE_SENSORS'
 static inline void AbiBindMsgRANGE_SENSORS(uint8_t sender_id, abi_event * ev, abi_callbackRANGE_SENSORS cb) {
                                                                               ^
/home/knmcguire/paparazzi/var/include/abi_messages.h:378:73: error: expected ';', ',' or ')' before 'distance'
 static inline void AbiSendMsgRANGE_SENSORS(uint8_t sender_id, int16_t[] distance) {
                                                                         ^
CC /home/knmcguire/paparazzi/var/aircrafts/ardrone2_rangesensors_gazebo/nps/state.o
CC /home/knmcguire/paparazzi/var/aircrafts/ardrone2_rangesensors_gazebo/nps/firmwares/rotorcraft/main.o
CC /home/knmcguire/paparazzi/var/aircrafts/ardrone2_rangesensors_gazebo/nps/firmwares/rotorcraft/main_ap.o
/home/knmcguire/paparazzi/conf/Makefile.nps:97: recipe for target '/home/knmcguire/paparazzi/var/aircrafts/ardrone2_rangesensors_gazebo/nps/subsystems/settings.o' failed
make[1]: *** [/home/knmcguire/paparazzi/var/aircrafts/ardrone2_rangesensors_gazebo/nps/subsystems/settings.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /home/knmcguire/paparazzi/sw/airborne/subsystems/abi.h:32:0,
                 from firmwares/rotorcraft/main_ap.c:75:
/home/knmcguire/paparazzi/var/include/abi_messages.h:60:72: error: expected ';', ',' or ')' before 'distance'
 typedef void (*abi_callbackRANGE_SENSORS)(uint8_t sender_id, int16_t[] distance);
                                                                        ^
/home/knmcguire/paparazzi/var/include/abi_messages.h:371:79: error: unknown type name 'abi_callbackRANGE_SENSORS'
 static inline void AbiBindMsgRANGE_SENSORS(uint8_t sender_id, abi_event * ev, abi_callbackRANGE_SENSORS cb) {
                                                                               ^
/home/knmcguire/paparazzi/var/include/abi_messages.h:378:73: error: expected ';', ',' or ')' before 'distance'
 static inline void AbiSendMsgRANGE_SENSORS(uint8_t sender_id, int16_t[] distance) {
                                                                         ^

gautierhattenberger, 2017-10-11:
ABI parser is not that smart, try to replace your type `int16_t[]` by `int16_t *` and send an extra parameter for the size of the array (I suggest at least an `uint16_t`)

knmcguire, 2017-10-13:
Just now I was able to check this, and manually adding the array size does seem to do the trick. Thank you for your quick responds

closed 2017-10-13 by knmcguire