paparazzi/paparazzi #2134: NPS: AHRS and INS are not bypassed during sensor callbacks, producing incorrect results

tomvand, 2017-10-24:
The current implementation of the AHRS and INS bypass results in incorrect state information in several sensor ABI message callbacks, among which the IMU_ACCEL message.

In [`nps_autopilot_rotorcraft.c:nps_autopilot_run_step(...)`](https://github.com/paparazzi/paparazzi/blob/master/sw/simulator/nps/nps_autopilot_rotorcraft.c#L93) the sensors are updated ([line 105-148](https://github.com/paparazzi/paparazzi/blob/master/sw/simulator/nps/nps_autopilot_rotorcraft.c#L105)) before the AHRS and INS are overwritten in line [150-156](https://github.com/paparazzi/paparazzi/blob/master/sw/simulator/nps/nps_autopilot_rotorcraft.c#L150).
The sensor updates send out the corresponding ABI messages, which causes the AHRS (and likely INS, not confirmed) to update its estimate and overwrite the vehicle state in state.h, even though it is disabled! As a result, the stateGetXXX functions return incorrect values when called from the sensor callbacks, as the vehicle state is only overwritten by its true value after all ABI callbacks have been completed.

I have not found a clean solution to this problem:
* Overwriting the vehicle state *beforehand* does not fix the problem as it does not overwrite the AHRS' internal state. Even if the AHRS' state could be overwritten, this would not prevent it from updating the vehicle state before the next subscribers are called.
* Both the AHRS and other modules subscribe to the sensor messages, which makes it difficult to update the modules but skip the AHRS.
* Setting the AHRS' sender ID's to ABI_DISABLE would prevent it from subscribing to the sensor messages, but 1) the bypass of the AHRS can be changed at runtime and 2) the sender ID define names differ per AHRS implementation.
* Sensor message callbacks are not shared between the AHRS implementations, so there is no central place to check the bypass flag when new sensor values arrive at the AHRS. Checking this flag in every AHRS implementation would solve this problem, but results in duplicate code and could be forgotten in new implementations. Similarly, there is no shared output function that overwrites the vehicle state in state.h that could check this flag.
* It looks like [ahrs.c](https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/ahrs.c#L101) *can* actually enable and disable AHRS. Perhaps this can be (ab)used to disable the AHRS? I could not find a similar mechanism for the INS though.

gautierhattenberger, 2017-10-24:
You are right that this will affect both AHRS and INS and that it is not a valid option to not broadcast the IMU data with ABI messages.
In order to keep the option to change the bypass flag at runtime, the only valid option is the last: use the `enable` functions to disable all registered AHRS so that only the bypass function is changing the state interface.
Several possible issues/required modifications:
- check that all AHRS are implementing it
- implement for INS since it is not done
- add `disable_all` and `restore_last` functions in AHRS/INS interfaces (might be done on NPS side as well, maybe this is even safer)

For a longer term option: integrate a selection in state interface input stage so that the enable/disable code is not duplicated in all estimation filter

tomvand, 2017-11-07:
I've fixed the bypass issue for the AHRS by implementing the `disable_all` and `restore_last` functions, see [here](https://github.com/paparazzi/paparazzi/compare/master...tomvand:fix_nps_bypass). The only AHRS not implementing this are ahrs_gx3.c and ahrs_sim.c, but as far as I'm aware these cannot be used with NPS anyway.

I'm now trying to do the same for the INS, but I'm not sure what to do with the `stateSetLocal(Utm)Origin` calls. NPS uses `stateSet*Ned` to overwrite the drone's state, so in order to get the correct results in global coordinates, the LTP origin needs to match the origin of the Gazebo world, right? Currently, Gazebo does not set the LTP origin at all, instead it seems that this is still set by the INS (using the drone's initial GPS position) even though it is bypassed. (This probably also explains why the drone needs some time to jump to the correct position in the GCS...)

I propose to fix this as follows: from `sim_overwrite_ins()` I could overwrite the local origin if it is not equal to Gazebo's world origin. When the INS is bypassed, this ensures that the drone's state is also correct in global coordinates; without the bypass both the LTP origin and drone's position are set from the simulated GPS coordinates, which should also produce the correct result.

The thing I'm not sure about is what happens when the INS no longer has full control over the local origin:
1. *What should happen when the INS attempts to set the local origin while it is bypassed?* I'm not exactly sure what the use case is of functions like `ins_reset_local_origin`, but when the INS is bypassed it cannot overwrite the local origin and that function would fail. Would it be ok to just return from this function without doing anything, or does that break some other part of the code?
2. What happens when the local origin is overwritten by Gazebo and then the INS bypass is turned off? *Can all INS handle an external change of the local origin?* Since the `ins_reset_local_origin` functions do little more than calling `stateSetLocalOrigin`, they probably can; as long as the INS gets its current state using `stateGet*` it should work fine. However, for instance `ins_int.c` also seems to maintain its own position estimate (`ins_int.ltp_pos`) which might become invalid if the local origin is changed. I do not know the INS implementations well enough to determine whether this will be a problem or not.

flixr, 2017-11-09:
Different idea: now that we (for the most part) support multiple AHRS and INS we could also just create simulated versions that always give you the true values from the simulator (with possibly added noise).
You should be able to select which one you actually to populate the state interface at runtime just as if you had multiple "normal" ones.
IMHO that is at least what we should go with in the long run...

tomvand, 2017-11-10:
Hmm I really like the idea of having a sort of 'ground-truth' AHRS/INS. I can try to implement this for the AHRS and see how it works, but as far as I'm aware there is no switching mechanism for the INS yet. Given that you already worked on this before (#1145), would you have time to look at this?

flixr, 2017-11-13:
Unfortunately I currently don't really have much time ;-( so don't expect something from me on that front soon...