paparazzi/paparazzi #2138: bebop2 upload failed

duguangxun, 2017-10-25:
Code v5.12 stable
Upload details:
-------------------------------------------------------------------------------------------------------------------
RUN 'make -C /home/adu/paparazzi -f Makefile.ac AIRCRAFT=bebop2  ap.upload '
make: Entering directory `/home/adu/paparazzi'
-----------------------------------------------------------------------
Paparazzi version v5.12_stable-2-g9c592a7-dirty
-----------------------------------------------------------------------
#######################################
# BUILD AIRCRAFT=bebop2, TARGET ap
#######################################
make[1]: Entering directory `/home/adu/paparazzi'
make[1]: Nothing to be done for `radio_ac_h'.
make[1]: Leaving directory `/home/adu/paparazzi'
make[1]: Entering directory `/home/adu/paparazzi'
make[1]: Nothing to be done for `flight_plan_ac_h'.
make[1]: Leaving directory `/home/adu/paparazzi'
make[1]: Entering directory `/home/adu/paparazzi'
GENERATE autopilots in /home/adu/paparazzi/var/aircrafts/bebop2/ap/generated/
GENERATE /home/adu/paparazzi/var/aircrafts/bebop2/ap/generated/settings.h
make[1]: Leaving directory `/home/adu/paparazzi'
Info: module 'imu_common.xml' has been loaded several times for target ap, merging options
Info: module 'gps.xml' has been loaded several times for target ap, merging options
Info: module 'imu_common.xml' has been loaded several times for target nps, merging options
Info: module 'gps.xml' has been loaded several times for target nps, merging options
cd sw/airborne; make -j8 TARGET=ap all
make[1]: Entering directory `/home/adu/paparazzi/sw/airborne'

Using CC   = arm-linux-gnueabi-gcc
CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/subsystems/settings.o
GCC version:
CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/autopilot.o
arm-linux-gnueabi-gcc (Ubuntu/Linaro 4.7.3-12ubuntu1) 4.7.3

CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/firmwares/rotorcraft/autopilot_static.o
CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/subsystems/gps.o
CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/subsystems/datalink/datalink.o
LD /home/adu/paparazzi/var/aircrafts/bebop2/ap/ap.elf
make[1]: Leaving directory `/home/adu/paparazzi/sw/airborne'
cd sw/airborne; make TARGET=ap upload
make[1]: Entering directory `/home/adu/paparazzi/sw/airborne'
Traceback (most recent call last):
  File "/home/adu/paparazzi/sw/tools/parrot/bebop.py", line 53, in <module>
    bebop.parse_args()
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 372, in parse_args
    self.upload_and_run(args.file, args.folder, args.min_version, args.max_version)
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 277, in upload_and_run
    v = self.check_version()
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 219, in check_version
    return ParrotVersion(self.execute_command('cat ' + self.version_file).strip())
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 55, in __init__
    self.__init__()
TypeError: __init__() takes exactly 2 arguments (1 given)
make[1]: *** [upload] Error 1
make[1]: Leaving directory `/home/adu/paparazzi/sw/airborne'
make: *** [ap.upload] Error 2
make: Leaving directory `/home/adu/paparazzi'

FAILED 'make -C /home/adu/paparazzi -f Makefile.ac AIRCRAFT=bebop2  ap.upload' with code 2
---------------------------------------------------------------------------------------------------------------------------

Can anyone help me out with this?

gautierhattenberger, 2017-10-25:
can you try to change the line 55 of the file `paparazzi/sw/tools/parrot/parrot_utils.py` to have `self.init(self)` and report the result ? Thanks

duguangxun, 2017-10-26:
@gautierhattenberger Thanks for  your reply. I have changed the line 55 of the file paparazzi/sw/tools/parrot/parrot_utils.py to have self.init(self) and the results are as follows:
-----------------------------------------------------------------------------------------------------------------------------------------
RUN 'make -C /home/adu/paparazzi -f Makefile.ac AIRCRAFT=bebop2  ap.upload '
make: Entering directory `/home/adu/paparazzi'
-----------------------------------------------------------------------
Paparazzi version v5.12_stable-2-g9c592a7-dirty
-----------------------------------------------------------------------
#######################################
# BUILD AIRCRAFT=bebop2, TARGET ap
#######################################
make[1]: Entering directory `/home/adu/paparazzi'
make[1]: Nothing to be done for `radio_ac_h'.
make[1]: Leaving directory `/home/adu/paparazzi'
make[1]: Entering directory `/home/adu/paparazzi'
make[1]: Nothing to be done for `flight_plan_ac_h'.
make[1]: Leaving directory `/home/adu/paparazzi'
make[1]: Entering directory `/home/adu/paparazzi'
GENERATE autopilots in /home/adu/paparazzi/var/aircrafts/bebop2/ap/generated/
GENERATE /home/adu/paparazzi/var/aircrafts/bebop2/ap/generated/settings.h
make[1]: Leaving directory `/home/adu/paparazzi'
Info: module 'imu_common.xml' has been loaded several times for target ap, merging options
Info: module 'gps.xml' has been loaded several times for target ap, merging options
Info: module 'imu_common.xml' has been loaded several times for target nps, merging options
Info: module 'gps.xml' has been loaded several times for target nps, merging options
cd sw/airborne; make -j8 TARGET=ap all
make[1]: Entering directory `/home/adu/paparazzi/sw/airborne'

Using CC   = arm-linux-gnueabi-gcc
CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/subsystems/settings.o
GCC version:
CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/autopilot.o
CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/firmwares/rotorcraft/autopilot_static.o
arm-linux-gnueabi-gcc (Ubuntu/Linaro 4.7.3-12ubuntu1) 4.7.3

CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/subsystems/gps.o
CC /home/adu/paparazzi/var/aircrafts/bebop2/ap/subsystems/datalink/datalink.o
LD /home/adu/paparazzi/var/aircrafts/bebop2/ap/ap.elf
make[1]: Leaving directory `/home/adu/paparazzi/sw/airborne'
cd sw/airborne; make TARGET=ap upload
make[1]: Entering directory `/home/adu/paparazzi/sw/airborne'
Traceback (most recent call last):
  File "/home/adu/paparazzi/sw/tools/parrot/bebop.py", line 53, in <module>
    bebop.parse_args()
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 372, in parse_args
    self.upload_and_run(args.file, args.folder, args.min_version, args.max_version)
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 277, in upload_and_run
    v = self.check_version()
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 219, in check_version
    return ParrotVersion(self.execute_command('cat ' + self.version_file).strip())
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 55, in __init__
    self.__init__(self)
  File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 55, in __init__
    self.__init__(self)
.
.
.
.then repeat the above sentence "File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 55, in __init__
    self.__init__(self)" 100 times and even more
.
.
.
 File "/home/adu/paparazzi/sw/tools/parrot/parrot_utils.py", line 55, in __init__
    self.__init__(self)

RuntimeError: maximum recursion depth exceeded
make[1]: *** [upload] Error 1
make[1]: Leaving directory `/home/adu/paparazzi/sw/airborne'
make: *** [ap.upload] Error 2
make: Leaving directory `/home/adu/paparazzi'

FAILED 'make -C /home/adu/paparazzi -f Makefile.ac AIRCRAFT=bebop2  ap.upload' with code 2
-------------------------------------------------------------------------------------------------------------------------

It seems that the file paparazzi/sw/tools/parrot/parrot_utils.py has some bugs. 
Thanks!

gautierhattenberger, 2017-10-27:
sorry I did it wrong, it should be `self.__init__()` in line 55. This is fixed in the master branch https://github.com/paparazzi/paparazzi/blob/master/sw/tools/parrot/parrot_utils.py#L55

gautierhattenberger, 2017-10-27:
Actually this is really not done correctly, we only need 1 init function can you try to replace the 2 init function by this piece of code ?
```
    def __init__(self, s):
        try:
            self.raw = s
            ss = s.split(".")
            self.h = int(ss[0])
            self.m = int(ss[1])
            sss=ss[2].split("-RC")
            self.l = int(sss[0])
            if len(sss) > 1:
                self.rc = int(sss[1])
            else:
                self.rc = 0
        except:
            self.h = 0
            self.m = 0
            self.l = 0
            self.rc = 0
            self.raw = ''
```

kirkscheper, 2017-11-28:
@duguangxun any update? Did the suggested fix work?

commit 86101eca (2017-12-14): [parrot] fix upload on recent bebop2

Applying the modification proposed in #2138 after testing it.
The version check is for now disabled if the correct valid cannot be
retrive from the drone. This should be improved later.

close #2138
closed 2017-12-14 by gautierhattenberger