paparazzi/paparazzi #2140: Deprecated Warnings for Ocaml ≥ 4.04

rijesha, 2017-10-25:
Ubuntu 17.10 ships with ocaml v4.04.0

Strings.capitalize has become String.capitalize_ascii
Strings.lowercase has become String.lowercase_ascii

deprecated: Scanf.fscanf
Use Scanning.from_channel then Scanf.bscanf.

Warning 52: Code should not depend on the actual values of
this constructor's arguments. They are only for information
and may change in future versions. (See manual section 8.5)

A similar issue for ocaml ≥ 4.02
https://github.com/paparazzi/paparazzi/issues/1703

rijesha, 2017-12-16:
@gautierhattenberger  Do you have a preference on how to fix this? I can do the same thing that I did last time with 4.02. The compat module will get a lot more complex.

gautierhattenberger, 2017-12-16:
I don't have a better option for `String` for now. For `Scanf` it depends if the alternate option is supported by other 4.0x versions or not.

BTW, do you compile with ocaml from Ubuntu packages or from opam ? It seems that system packages have an issue with ocamlbuild. Do you have that as well ?

rijesha, 2018-01-25:
Very late reply, but recently I have not had any issues with system or opam versions of ocaml

closed 2018-01-26 by rijesha