paparazzi/paparazzi #2144: VELOCITY_ESTIMATE ABI message requires vertical velocity

tomvand, 2017-10-30:
With the current definition of the VELOCITY_ESTIMATE ABI message it is not possible to send only a horizontal velocity estimate, the vertical velocity also has to be included. Setting the vertical velocity estimate to zero or stateGetSpeedNed_f()->z as a workaround results in incorrect behaviour. Can the message be changed to include a flag that indicates which axes are estimated, or separated into a horizontal and vertical estimate?

gautierhattenberger, 2017-10-30:
A flag can be added similar to https://github.com/paparazzi/pprzlink/blob/master/message_definitions/v1.0/messages.xml#L674 in order to specify if horizontal and/or vertical estimate have been updated

kirkscheper, 2017-11-28:
closed with #2161

closed 2017-11-28 by kirkscheper