paparazzi/paparazzi #2153: Fixedwing and rotorcraft and flightplan primitives not universal

OpenUAS, 2017-10-31:
As user I would like to have a flightplan that is usable for as well FW and Rotorcraft mode, A.k.a hybrid vehicle. For this one needs universal fixedwing and rotorcraft and flightplan **primitives** as a first step. This step would universalizes current flightplan syntax more so fixwing and rotorcraft are unified in primitive syntax. Greater goal: Universal Rotorcraft and Fixedwing... 

Stay: The issue with hybrid, 3D prophanger, Quadshot, Cyfoam and the likes

The _S_ word...: "Stay"

A) Stay with fixedwing is like flowering through the "stay" waypoint
B) Stay with rotorcraft is hovering at that waypoint

But...prop hangers, like 3D Stunt planes, Cyfom, Quadshot, X-vert and others **can do both**! So it is undetermined what exact behavior in the flightplan one wants

# Solution

Introduce new primitive used in forward mode which does the same thing as the current fixedwing "Stay" command

# Find good primitive term

**Stay** synonym to choose from: _abide, bide, remain, dwell, linger, loiter, dwell_ - To continue to be in a place: Read more at http://thesaurus.yourdictionary.com/bide#6x2ZVEeWzeLtpGh8.99

The chosen term will be : "**abide**" behaviour descriptive correct and different enough from "stay". Note that We chose _not to use linger_ in UAS that is well know to be used for hanging around mostly in a circular pattern...

NOTE:
* For this nothing changes for rotorcraft users (100% of users happy..)
* For FW only Stay can still do the same (99% of users happy)
* Hybrid, we just better deterministic flightplans.. (101% of users happy ;)

OpenUAS, 2017-10-31:
Assigned to self to work on it... Please add comment, or even better code or solution as a :gift:

gautierhattenberger, 2017-11-07:
I'm not really sure to see the issue here. Do you really want to have a hybrid fly over and over above a stay waypoint without hovering ?
I've always think that using `stay` for a plane should be avoid and use `circle` instead as the trajectory is much more predictable. If you need to fly over the point, then you can use the `eight` pattern.
If you really want to force a hybrid type plane to fly over instead of hovering, I think you should put a speed constraint rather than adding a flight plan pattern. If you have a guidance function to set this minimum value, you can use the `pre_call` or `post_call` to set it.
Otherwise, adding some more attributes like speed and heading constraints on waypoints was already discussed/requested. I think it is a good idea but it needs to redesign the nav API as it does not comply with this.
And redesigning nav is a long and painful but necessary task anyway.

noether, 2018-03-06:
I agree with Gautier. However, I was thinking whether there are any plans for redesigning the nav API. In particular, I was thinking about the integration of the GVF in the nav functions. It is a pity that now we cannot use the gvf properly, but by calling the corresponding c function in the flightplan.

I do not know whether is a good idea just to add gvf_nav functions (so we will have a duplicated set), or even to leave the nav api untouched, and to add an extra layer in the code, so the user can choose carrot/gvf/others as a guidance system in the airframefile (leaving the carrot by default for the older aircraft).

OpenUAS, 2018-07-17:
> Do you really want to have a hybrid fly over and over above a stay waypoint without hovering ?
Indeed, energy an string wind wise, e.g. Delftacopter
For now since self assigned and More hybrid flight with all kinds of scenarios coming up

gautierhattenberger, 2018-12-05:
I take this out of v5.14 as their is not short term plan to change nav API. But an option to improve things would be to have a look at the rover firmware that allows to attach different kind of nav/guidance loop behind the flight plan primitives.

OpenUAS, 2018-12-18:
OK. Thx for good hint

OpenUAS, 2018-12-18:
BTW we flew auto2 the Disco on Hybrid Delftacopter code a while ago. Just letting FW "rust" and focusing on adding more FW into ROTORCRAFT is also an option.