paparazzi/paparazzi #2166: Discussion about secured datalink

gautierhattenberger, 2017-11-06:
This thread is a discussion regarding the choices and implementations of a secured protocol for PPRZLINK. Several things have already been done by @podhrmic with #2162  #2163 #2164 #2165 and the wiki page https://wiki.paparazziuav.org/wiki/Messages_Format#Secure_Paparazzi_.28S-PPRZ.29_for_serial_and_transparent_modem_communication

Several issues and questions arise from this:
- do we need asymmetric keys and a exchange protocol or a symmetric key flashed and compiled with the AP code is enough ?
- do we need a key or pair of keys for each aircraft or a single one for all of them ?
  - generating a single key or pair could be done once when building the project
  - a single key should make things easier for air-to-air communication
  - since aircraft are flashed by the GCS (most of the time with a USB cable) it should be safe
  - @podhrmic any particular reason to use different keys for each aircraft ?
- which encryption algorithm(s) should be implemented/supported (ed25519, aes, ...) ?
- which languages/bindings should be involved (python, ocaml, rust) on the ground side ?
- where should be installed/packaged the encryption libraries (sw/airborne, sw/ext) ?

I will let @podhrmic  develop the solution he proposed. Here is a short insight of a preliminary work made at Enac regarding this matter:
The idea was to guarantee that every message is consistent, unique and safe. First point is already done by the checksum. The second can be handled by a timestamp/counter (inside the encrypted part of the message) preventing an attack by "replaying" the messages. The last point, the encryption itself (that guarantees both authenticity and secrecy) was using a standard aes encryption based on mbedtls (https://tls.mbed.org/, https://github.com/ARMmbed/mbedtls) for the airborne part and openssl lib for the ground part. One of the reason is that bindings already exist for python and ocaml. The key is generated once when building the project and can be included in every airframe compiled and flashed by the ground station.
There is a small overhead with the timestamp, the encryption crc, actual payload size and the fact that encryption is done on fixed size blocks. Also, as mentioned in https://github.com/paparazzi/pprzlink/pull/65#discussion_r148921313, encryption is done when all the sending buffer is ready. For bare-metal implementation, an extra intermediate buffer shall be used. For ChibiOS implementation, an other option is to use the dynamic allocation with tlsf (already used for sdlog).
Some other questions: if we want to use this with an other transport like XBee, do we need several specific implementation or can we make this adaptable for any transport ? With PPRZLINK v2, the sender and receiver ID (part of the payload) might be used for routing, how do we handle it ?

Last remark (but not least): Paparazzi is about experimenting new solutions, so it is not an issue to support several options. In this regard, I suggest that the modules names include the encryption type.

podhrmic, 2017-11-06:
Thanks for opening this issue @gautierhattenberger, it makes the discussion easier.

To your comments:

> do we need asymmetric keys and a exchange protocol or a symmetric key flashed and compiled with the AP code is enough ?

The key exchange with asymmetric keys is required to avoid using the same symmetric key with the same initialization vector (IV or also _nonce_ or in this case _counter_) twice. If we encrypt two different messages with the same key and the same IV, the encryption algorithm doesn't hold its security properties anymore (a real-world example of this is the recent Wifi WPA2 KRACK vulnerability  https://www.wired.com/story/krack-wi-fi-wpa2-vulnerability/ ). Without going into too much detail, using a unique IV over multiple session on embedded targets is hard (a way how to avoid that was mentioned in my [recent paper](http://ieeexplore.ieee.org/document/7991460/)), hence the common key exchange protocol. 

> do we need a key or pair of keys for each aircraft or a single one for all of them ?

In the current scenario I assume only one-to-one communication, with the possible extension for many-to-many later.

> generating a single key or pair could be done once when building the project

True, but each airplane needs its own key (the GCS key can be shared). That way the messages can be verified to come from a specific airplane.

> a single key should make things easier for air-to-air communication

Also true, but makes it less secure. As mentioned above, it is hard to keep track of the IVs that were used, especially in many-to-many communication. 

> since aircraft are flashed by the GCS (most of the time with a USB cable) it should be safe

Yes, the code upload is considered to be trusted (the UAV has to know the GCS public key before the key exchange can begin).

> any particular reason to use different keys for each aircraft ?

Two: first, to be able to verify that given message comes from a specific aircraft, second: to avoid accidentally reusing the IV

> which encryption algorithm(s) should be implemented/supported (ed25519, aes, ...) ?

ed25519, curve25519 and SHA512 are required for the key exchange, Chacha20 with Poly1305 authenticator is used for encrypted communication. Chacha is faster than AES on purely software implementation while having comparable security properties, so it is good (and somewhat recommended choice) for multiplatform applications (like Paparazzi with multiple different autopilots). AES is faster in hardware, but I don't want to be writing AES drivers for each board. Also, the hardware implementation is typically closed source and hence cannot be independently verified:-(

> which languages/bindings should be involved (python, ocaml, rust) on the ground side ?

Ideally OCaml, second is Rust. The simple fact is that I don't know OCaml, but I know Rust. Rust is memory safe and fast, so it is a good choice for such a critical piece of infrastructure as is the GCS link program. I already have HACL bindings for Rust, as well as pprzlink in Rust. I know it is another dependency, but perhaps it is justifiable price for the added security.

> where should be installed/packaged the encryption libraries (sw/airborne, sw/ext) ?

HACL-C is fairly low footprint (some 20 files or so), so it made sense to keep it as a module. Also, the HACL* developers don't have a separate git repo for the C snapshots, so we would have to pull the whole HACL* repo which is quite large. I am OK with changing that later if needs to be.

> There is a small overhead with the timestamp, the encryption crc, actual payload size and the fact that encryption is done on fixed size blocks. 

From my preliminary tests, the overhead on F4 autopilots is small enough to be negligible (less than 10% CPU). Indeed, having a separate low priority thread for the encryption/decryption would be ideal, but that is a different story.

> Some other questions: if we want to use this with an other transport like XBee, do we need several specific implementation or can we make this adaptable for any transport ?

I haven't looked into it, but in general the encryption can be added to other transports as well (at which point we should factor it out). For now though, I would prefer sticking with the transparent link only.

> With PPRZLINK v2, the sender and receiver ID (part of the payload) might be used for routing, how do we handle it ?

I decided to not to encrypt the sender/receiver IDs (but still authenticate them), so the packets can be freely routed/filtered, but decrypted only when needed.

> Last remark (but not least): Paparazzi is about experimenting new solutions, so it is not an issue to support several options. In this regard, I suggest that the modules names include the encryption type.

The secure link uses a modification of Galois Embedded Crypto that we developed for SMACCMPilot. The main difference for pprzlink that I propose is to use a formally verified crypto lib HACL, instead of other more common (yet unverified and potentially buggy) libraries (mbed, libsodium etc.). To my knowledge, a verified crypto library hasn't been used for UAV communication before, which is good news for paparazzi:-)

## Algorithm
For ongoing communication, we use Chacha20 with Poly1305 auhenticator. That allows the user to send encrypted payload, as well as plaintext data that are authenticated. What it means is that if an adversary changes the data (for example _sender ID_), the authenticator will notice it and not decrypt the modified message. That way the integrity of also unencrypted data is protected.

The replay attacks are prevented by monotonically increasing counter, which serves also as IV for each message (hence a unique IV is quaranteed).

OpenUAS, 2017-11-07:
It is a bit unclear how multilink is handled, have to admit have still to test this fantastic addition. A;though IMHO an AES option is a must to interact with solution where we do not have much choice etc

podhrmic, 2017-11-07:
I ll refactor the algorithm to make it easier to employ different versions of encryption. That way the pprzlink itself will only call generic encrypt/decrypt functions, and all the specifics will be handled in a Paparazzi module.

Any thoughts on introducing Rust (as in #2165 )? Maybe we can have Rust programs for now (to have a working example), and in time people might want to extend/write OCaml/Python equivalents?

gautierhattenberger, 2018-04-12:
closed with the support of the secured version of pprzlink

closed 2018-04-12 by gautierhattenberger