paparazzi/paparazzi #2167: Set PPRZLINK v2.0 as default

gautierhattenberger, 2017-11-06:
During the last developer meeting, one question was about using PPRZLINK v2 as default as it should be transparent for the users and it offers more functionalities.
I said that if we manage to fly a formation of UAV with air-to-air communications during the IMAV2017 competition it would be good enough to make the transition.
And we did it :) 
https://blog.paparazziuav.org/2017/08/12/multi-uav-capabilities/
http://www.imavs.org/papers/2017/217_imav2017_proceedings.pdf

So, what do you guys think ? Should we upgrade for version 2 now ?

dewagter, 2017-11-06:
I prefer upgrading, which will 1) yield more testing and 2) make it
standard for any new developments rather sooner than later.

-Christophe

podhrmic, 2017-11-06:
Fine with me *thumbs up*

hooperfly, 2017-11-07:
As part of the transition, will there be an documentation that discusses upgrading from PPRZLink v1.0 to v2.0? The pitfalls, recommendations, examples, etc...?

gautierhattenberger, 2017-11-07:
that would be indeed a nice thing to have a documentation...

OpenUAS, 2017-11-07:
Thanks for all the hard work everyone, v2.0 bring it on 👍

flixr, 2017-11-07:
I haven't had the time to really look at it, but to me it sounds like it would be good to make it the default (as long as we have some docs for it).

gautierhattenberger, 2017-11-24:
Is this good enough ? https://wiki.paparazziuav.org/wiki/Pprzlink

podhrmic, 2017-11-27:
@gautierhattenberger LGTM. I think the docs are good enough (== I can understand how to use pprzlink 2.0).

closed 2017-11-29 by podhrmic