paparazzi/paparazzi #2181: Position jump to HOME when modules section is moved to firmware section

dewagter, 2017-11-09:
@guidoAI can you please tell which airframe and

guidoAI, 2017-11-10:
Hi, I have a bebop 2 airframe file with front vision. I worked with a master version of ~1-2 weeks ago.

In the airframe file I first had the gps datalink only under firmware, since I heard that now could / should be done:

<firmware name="rotorcraft">
    <target name="ap" board="bebop2"/>
  ...
    <module name="gps" type="datalink"/> 
  ...
  </firmware>

I did NOT have a gps under modules:

<modules main_freq="512">
...
</modules>

When flying with the motion tracking system, the position estimate would jump from the actual location to (0,0,0) (not home, as when I moved that it kept jumping to the center of the flight arena). 

Adding the gps under the modules part solved the problem. Also when removing it from the firmware part.

gautierhattenberger, 2017-11-13:
Something that may change is the order of loading of the modules. You can check that in `var/aircrafts/<AC_NAME>/<TARGET>/generated/modules.h`. @guidoAI can you compile the aircraft in both configuration and report the content of modules generated init ?

gautierhattenberger, 2021-03-25:
The whole module system is being changed, so if this is still an issue in mster, adding proper dependency between modules might solve the problem.
The `modules` section will be removed in pprz v6.x anyway. Please open this again with more information on the loaded modules if it is still a problem.

closed 2021-03-25 by gautierhattenberger