paparazzi/paparazzi #2182: WARNING on all sim targets: /usr/lib/ocaml: linker input file unused because linking not done

dewagter, 2017-11-10:
e.g. test_conf with twog_imu:

GENERATE /home/mavlab/paparazzi/var/aircrafts/Twog_IMU/sim/generated/settings.h
../../sw/tools/find_vpaths.py gcc /home/mavlab/paparazzi/var/aircrafts/Twog_IMU/sim_tmp.list /home/mavlab/paparazzi
gcc: warning: /usr/lib/ocaml: linker input file unused because linking not done
make[1]: Leaving directory `/home/mavlab/paparazzi'

gautierhattenberger, 2017-11-10:
it looks like this vpath script is causing a lot of issues. What is it's real use anyway ? Isn't their a better way to extract the dependencies of the C/C++ files during normal compilation rather than running separately with incorrect parameters ?

kevindehecker, 2017-11-10:
That script became necessary because the vpaths broke the qt project generator some while back. The script finds the full paths of all source files for the qt project generator. 

I like it to remain functional because Qt shows only the used source files. Once you get used to that there is no turning back... You can easily jump through the code, as there are no ambiguities anymore. (caused by multiple function implementations in different files)

At least one problem is that the arguments put into gcc -m in the qt generator are different than the final arguments, because in later makefiles even more arguments are augmented. Haven't looked for a good way around this yet, but the build system of paparazzi is so outrageously complex...

What's the real use of these vpaths btw?

podhrmic, 2017-11-29:
I fixed #2194 with 061c38981155fa0574e8affdfca7a26fe8670b4c but it can be improved. Looking at this, what about calling the `find_vpaths.py` only if you are compiling from Qt? (you could just pass a flag to the makefile). @kevindehecker @gautierhattenberger ?

podhrmic, 2017-11-29:
To add to the discussion - the python script seems to obfuscate errors - for example if I am trying to link a non-existing file, I get this output:

```
./../sw/tools/find_vpaths.py arm-none-eabi-gcc /home/michal/Documents/Paparazzi/tmp/paparazzi/var/aircrafts/Quad_LisaMX/ap_tmp.list /home/michal/Documents/Paparazzi/tmp/paparazzi
arm-none-eabi-gcc: error: ERROR:: No such file or directory
arm-none-eabi-gcc: error: could: No such file or directory
arm-none-eabi-gcc: error: not: No such file or directory
arm-none-eabi-gcc: error: find: No such file or directory
arm-none-eabi-gcc: error: src:: No such file or directory
arm-none-eabi-gcc: error: modules/datalink/hacl-c/Chacha20.cERROR:: No such file or directory
```
You can see that 5 lines of the error output are extra. I might have a look at it later, but I am not sure what is the end goal here (like what exactly do you need for the Qt to build).

kevindehecker, 2017-11-30:
You cannot easilily compile from qtc if there are no project files generated in the first place, which currently happens during building. I think the best thing to do is to add a new menu item "generate QT project", and remove this step from the build process. There is no need to regenerate the project every build anyway.  (tried adding it to the new Python PPRZ center real quick, but currently can't get it to start)

OpenUAS, 2018-12-04:
Still an issue?

gautierhattenberger, 2019-11-15:
`vpath` is now limited to projects where it makes sense.
closing this

closed 2019-11-15 by gautierhattenberger