paparazzi/paparazzi #2185: Battery monitor shows inconsistent cell voltages

podhrmic, 2017-11-15:
[Battery monitor module](https://github.com/paparazzi/paparazzi/blob/master/conf/modules/battery_monitor.xml) shows fluctuating voltages.

[image]

Tested on bare-metal, Lisa MX.

Possible driver issue, making a ticket for myself

OpenUAS, 2018-02-01:
Any chance to peek or get or make such a custom board? If the issue is not self assigned willing but IMHO  no one else can fix this but oneself. Also to little info to give some hints, e.g. which bus connected etc.

podhrmic, 2018-02-01:
I will double check it, thanks for asking.

podhrmic, 2018-02-02:
@OpenUAS apparently the university has some weird ideas about intellectual property, so the custom hardware is not available for release at the moment, sorry:-( I ll try to improve the comments, the code is IMHO general enough that it can be easily adapted to any other board using AD7997-8 chips.

podhrmic, 2018-04-16:
Closing as this was apparently a hardware issue (so not a problem anymore).

closed 2018-04-16 by podhrmic