paparazzi/paparazzi #2187: Segfault in cv_async_function() when video changes size

tomvand, 2017-11-16:
I'm getting a segfault when my module attempts to change the resolution of the images returned from its video callback:

Asynchronous video listeners keep their own copy of the current video frame. This image is created in [`cv_async_function()` (cv.c line 114)](https://github.com/paparazzi/paparazzi/blob/d2079a780f316ef79c55ae30cc4e2e07a8c74144/sw/airborne/modules/computer_vision/cv.c#L114) with the size of the current frame `img`.
When new frames arrive they are copied to this image in [line 119](https://github.com/paparazzi/paparazzi/blob/d2079a780f316ef79c55ae30cc4e2e07a8c74144/sw/airborne/modules/computer_vision/cv.c#L119).

This fails when the size of the video frames changes between subsequent calls. [`image_copy()`](https://github.com/paparazzi/paparazzi/blob/d2079a780f316ef79c55ae30cc4e2e07a8c74144/sw/airborne/modules/computer_vision/lib/vision/image.c#L89) does not check that the size of the source image is the same as the target one, so when the new image is bigger, the memcpy on [line 89](https://github.com/paparazzi/paparazzi/blob/d2079a780f316ef79c55ae30cc4e2e07a8c74144/sw/airborne/modules/computer_vision/lib/vision/image.c#L89) writes beyond the allocated buffer and causes a segfault.

The problem can be fixed by re-creating the asynchronous copy of the image using the following code (inserted at line 116 of cv.c):
```
  if (async->img_copy.buf_size != img->buf_size) {
    printf("[cv_async_function] BUFFER SIZE CHANGED!\n");
    printf("Old:\t%d x %d\n", async->img_copy.w, async->img_copy.h);
    printf("New:\t%d x %d\n", img->w, img->h);
    image_free(&async->img_copy);
    image_create(&async->img_copy, img->w, img->h, img->type);
  }
```

Should this be fixed so that the size of the video can be changed, or should I not have attempted this in the first place? I could not find any documentation that this is not allowed. (Unless I missed something there is very little documentation on the video callback system of cv.c and video_thread.c).

Secondly, should image_copy check that the images' buffer sizes are equal before copying, so that segfaults like these are prevented or at least easier to find? Or should image_copy allocate a new buffer when the images' types or sizes are not equal?

kirkscheper, 2017-11-18:
I can answer the second part of the questions. The image copy should check that the buf_size of the destination is >= the origin. I would be ok with reallocating the image memory if it is too small.

For your code above, again I think it should be async->img_copy.buf_size **<** img->buf_size

For the rest I am not sure, maybe @dewagter has an idea.

dewagter, 2017-11-22:
Dear Tom,
This is the first case where video size changes as every camera driver has had hard-coded image sizes until now.
The solution to fix the buffer seems fine but only works as long as no cv_xxxx function makes a local copy of the image. All these must update their buffer as well.
So either we keep the fixed -- compile time -- efficient single image size constraint, or we must update many functions and make the project a little more complex.

tomvand, 2017-11-24:
I see your point, allowing the video size to change implies that the other functions/modules should be able to handle this, which is probably not the case right now and could add some unwanted complexity. Fixing the existing cv_xxx functions would also be a lot of work (checking the use of image_copy or image_create is easy enough, but determining whether the algorithms still work isn't). However, if the video size should remain fixed, then this should be clearly documented (maybe something for the upcoming paparazzi day?).
On the other hand, it can also be argued that this is a limitation of those cv functions and not of the video thread/streamer or my own module, both of which work fine with changing video sizes. So straight-up disallowing it while the cv_xxx functions are not even used is a bit limiting. The current cv_async_function() already allows the video size to change (as long as it does not exceed the initial size), so this fix would not drastically change its behavior.

Note that I currently only change the video size to show calibration information, so I can live with it if the video size really has to remain constant.

commit 829f8d77 (2018-05-30): unwrap_pano module for panoramic cameras (#2269)

* Create unwrap_pano module

Create basic settings structure.

* Get basic image unwrapping working

* Add module settings

* Fix incorrect declaration of pano_unwrap_init()

* Add video_thread dependency

* Fix incorrect minus sign on real drone

* Set default forward direction for ARDrone2

* Set default vertical resolution to 0.18

Seems to be a better match with the gazebo model.

* Use LUT for unwrapping

LUT seems to work fine in simulation, except it crashes when
overwrite_video_thread is set to false...?

* FIX Add workaround for overwr [... cut]
closed 2018-05-30 by OpenUAS