paparazzi/paparazzi #2194: Error when compiling Lisa_MX test airframe

podhrmic, 2017-11-27:
Using latest master and Quad_Lisa_MX from conf_tests.xml

```
GENERATE /home/michal/Documents/Paparazzi/paparazzi/var/aircrafts/Quad_LisaMX/ap/generated/settings.h
../../sw/tools/find_vpaths.py arm-none-eabi-gcc /home/michal/Documents/Paparazzi/paparazzi/var/aircrafts/Quad_LisaMX/ap_tmp.list /home/michal/Documents/Paparazzi/paparazzi
arm-none-eabi-gcc: error: ERROR:: No such file or directory
arm-none-eabi-gcc: error: found: No such file or directory
arm-none-eabi-gcc: error: more: No such file or directory
arm-none-eabi-gcc: error: than: No such file or directory
arm-none-eabi-gcc: error: once:: No such file or directory
arm-none-eabi-gcc: error: /home/michal/Documents/Paparazzi/paparazzi/sw/ext/libsbp/c/src/edc.c/home/michal/Documents/Paparazzi/paparazzi/sw/ext/libsbp/c/src/sbp.c: Not a directory
arm-none-eabi-gcc: error: ERROR:: No such file or directory
arm-none-eabi-gcc: error: found: No such file or directory
arm-none-eabi-gcc: error: more: No such file or directory
arm-none-eabi-gcc: error: than: No such file or directory
arm-none-eabi-gcc: error: once:: No such file or directory
arm-none-eabi-gcc: error: /home/michal/Documents/Paparazzi/paparazzi/sw/ext/libsbp/c/src/sbp.c/home/michal/Documents/Paparazzi/paparazzi/sw/airborne/arch/stm32/led_hw.c: Not a directory
```

When removing the piksi module, the problem goes away.

Any ideas? Both files (libsbp) exist as expected.

kirkscheper, 2017-11-27:
It looks like the piksi module uses absolute file extensions to find the c files. This doesn't work with sw/tools/find_vpaths.py since #1940. Maybe @kevindehecker can suggest a fix for you.

podhrmic, 2017-11-27:
OK, removing $(PAPARAZZI_SRC) didn't help. @kevindehecker any suggestions? I am surprised this didn't get caught by Travis:-/

commit 061c3898 (2017-11-27): Fixes #2194
commit 0d72c27d (2017-11-27): Fixes #2194
commit f79260fb (2017-11-28): Fixes #2194
closed 2017-11-27 by podhrmic