paparazzi/paparazzi #2203: Unfeasible timers not communicated

EwoudSmeur, 2017-12-05:
With ChibiOS, if timers (such as the periodic timer) are defined such that they take a non integer amount of clock ticks (SYS_TIME_FREQUENCY), the period is elongated to the nearest integer amount of clock ticks. https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/mcu_periph/sys_time.c#L51

Example: If you have PERIODIC_FREQUENCY = 512 and by default for ChibiOS SYS_TIME_FREQUENCY = 10000, the amount of clock ticks for this timer is 19.53, which becomes 20, so 10000/20 = 500Hz.

The real problem is that this is not communicated to the user. I would like to add a warning if this happens, but most of these calculations are done at run-time, and I don't think the C preprocessor can check if a number is an integer or not. Ideas?

Also, I am not sure if this applies if you do not use ChibiOS.

kirkscheper, 2018-01-29:
Is this fixed with #2224?

EwoudSmeur, 2018-01-30:
Yes, though maybe it is a good idea to do something similar for the other timers.

OpenUAS, 2018-07-31:
Should we close or do we need to create a new issue

OpenUAS, 2018-12-04:
Put on list at TU , a nice something for BS. Closed but not forgotten ;)

closed 2018-12-04 by OpenUAS