paparazzi/paparazzi #2207: Many implicit-fallthrough warnings triggered by GCC7

gautierhattenberger, 2017-12-20:
See https://developers.redhat.com/blog/2017/03/10/wimplicit-fallthrough-in-gcc-7/ for details about this type of error.

A (temporary) solution would be to add `-Wimplicit-fallthrough=0` apparently, but since this option is not recognized by older versions it needs to be filtered and used only with GCC7.
For long term solution, the code should be updated to add the proper comments.
Any other suggestions ?

podhrmic, 2017-12-28:
The `Label(x)` still triggers warning. I tried to fix it, but now I think that this switch + goto combination is something that GCC can't handle.

Getting rid of gotos in flightplan is out of question, at least for now. Perhaps we can switch to an older GCC? 

Here is the error I am getting:
```
paparazzi/var/aircrafts/Ark_Hexa_1.8/ap/generated/flight_plan.h: In function 'auto_nav':
./subsystems/navigation/common_flight_plan.h:45:1: warning: this statement may fall through [-Wimplicit-fallthrough=]
 nav_init_stage();
 ^~~~~~~~~~~~~~~~
./subsystems/navigation/common_flight_plan.h:57:16: note: in expansion of macro 'InitStage'
 { nav_stage++; InitStage(); }
                ^~~~~~~~~
paparazzi/var/aircrafts/Ark_Hexa_1.8/ap/generated/flight_plan.h:207:9: note: in expansion of macro 'NextStage'
         NextStage();
         ^~~~~~~~~
./subsystems/navigation/common_flight_plan.h:65:1: note: here
 label_ ## x:
 ^
paparazzi/var/aircrafts/Ark_Hexa_1.8/ap/generated/flight_plan.h:208:7: note: in expansion of macro 'Label'
       Label(while_1)
       ^~~~~
```

podhrmic, 2018-01-04:
@gautierhattenberger When you compile with ChibiOS it has own fail through warnings. Indeed it was fixed in the new versions, but that means we would have to update the repo. And that means redoing the tfls module. What do you think?

Also, we could revert to an older version of gcc?

gautierhattenberger, 2018-01-04:
We have planned to update ChibiOS when we have enough time to test things properly. In the meantime, they are only warnings. If it is blocking, using GCC6 should be good enough.

podhrmic, 2018-01-04:
Not blocking, just an observation. There is still the issue with 'Label' triggering the warning, the seems to be fine.

OpenUAS, 2018-01-24:
This is quite a good related read https://developers.redhat.com/blog/2017/03/10/wimplicit-fallthrough-in-gcc-7/

Using GCC 7.2 now

OpenUAS, 2018-01-24:
BTW not really the option to switch to older GCC that not to nice, since the warnings are valid warnings why hide from it? Using GCC 7.2.x without issue ATM (Except for ARDrone 2 that is.... *<:0)

kirkscheper, 2018-02-09:
should be closed with #2233

closed 2018-02-09 by kirkscheper