paparazzi/paparazzi #2222: Connection btw from computer to BMP to LİSA/S.

dynamaniac, 2018-01-25:
Hi all, 

we have a problem about connection btw computer and Lisa/S with BMP by using serial interface adapter. we did exactly what the procedure in wiki page said. However there is a problem states;

Target voltage; 0.0V
JTAG device scan failed!
/home/comp1/paparazzi/sw/tools/flash_scripts/bmp_jtag_flash.scr:9: Error in sourced command file:
Attaching to Remote target failed

Only things we did different from wiki page is using serial interface adapter instead serial wire debug cable because we couldnt plug that wire into BMP. 
Could anyone help about this issue?

OpenUAS, 2018-02-01:
Dear @dynamaniac You are fustrated about someting not working , very understandable.
However by asking this defect tracker of main code will unlikely help you.

It is better to use his:
https://gitter.im/paparazzi/discuss

Having sai that thre you better aslo
[ ] Could you add a picture orbe more decriptive where you connected cableing
[ ] Which BMP (v2.1?) you have
[ ] What upload method
[ ] how did you install PPRZ, specifically do you have the rules.d installed

tcunis, 2018-04-12:
Closed as `not a defect` until the opposite is demonstrated.

closed 2018-04-12 by tcunis