paparazzi/paparazzi #2223: Missing baro data in v5.12

HWal, 2018-01-28:
On my Lisa/M 2.0 with Aspirin 2.1, the Aspirin MS5611 baro is connected directly to Lisa/M via I2C.
I have <configure name="LISA_M_BARO" value="BARO_MS5611_I2C"/> in the airframe file.

This used to work in v5.10, but in v5.12 I get no baro data anymore.

I found that:
sw/airborne/boards/baro_board_ms5611_i2c.c

has the line:
#include "boards/bebop/baro_board.h

added in v5.12.

Commenting out the #include gives me baro data again.
Not sure if this is a permanent change, but is there an alternative position for this #include?

OpenUAS, 2018-02-06:
IMHO should be something along the lines of :
```
#include "subsystems/sensors/baro.h"
#if defined(BOARD_BEBOP) || defined(BOARD_DISCO)
#include "boards/bebop/baro_board.h"
#endif
#include "peripherals/ms5611_i2c.h"
#include "mcu_periph/sys_time.h"
#ifdef BARO_LED
#include "led.h"
#endif
```
Will need to test in BB B2 and Disco (And validate LIsa/M2 with aspirin 2.1 before a Disco pull request but can take a bit since need to testfly Disco first and rather not at -5 C ;) this week...

OpenUAS, 2018-04-04:
@podhrmic Is it thus the to backport this regression to v5.12 sonce you selfassigen? If so TIA ,if not ..ehh also fine ;-)

podhrmic, 2018-04-04:
@OpenUAS I am honestly not sure - but I assigned myself since it looks like a straightforward fix, just haven't had time yet. So feel free to take over:-)

OpenUAS, 2018-06-20:
OK, will do with @kirkscheper because of #1910

HWal, 2018-06-20:
Thank you for the help!

kirkscheper, 2018-07-25:
This should be solve in master now with #1910. Let use know if you still have any issues.

closed 2018-07-25 by kirkscheper

HWal, 2018-07-28:
Thanks for the help! I will test this as soon as I can.