paparazzi/paparazzi #2227: Semaphore CI failing

podhrmic, 2018-02-03:
Looks like it is a configuration error. Plus I don't think we need to build openCV when running Pull Request builds. I am not familiar with Semaphore CI so not sure how to fix it.

Perhaps we can even remove Sepahore and just use Travis?

Marking as blocking because it makes it hard to see if the PR is OK to merge.

@flixr @gautierhattenberger @kevindehecker what do you think?

The error message: 
```
make -C sw/ext opencv_bebop
git submodule init
git submodule update
cd opencv && git am --signoff < ../fix_compiler_crash.patch && cd ..

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'pprz@24873580ed7d.(none)')
Makefile:24: recipe for target 'all' failed
make[2]: *** [all] Error 128
Makefile:94: recipe for target 'opencv_bebop.build' failed
make[1]: *** [opencv_bebop.build] Error 2
Makefile:176: recipe for target 'opencv_bebop' failed
make: *** [opencv_bebop] Error 2
```

flixr, 2018-02-04:
So I see three possibilities:
1. simply set a name and email in the semaphore script
2. convert the git patch to a "regular" patch
3. create a fork of opencv with the patch and point the submodule to that

Without looking at the patch at all, I wonder why this hasn't been upstreamed though...

podhrmic, 2018-02-04:
Thanks for looking at it. Can you point me at the semaphore script?

flixr, 2018-02-04:
https://semaphoreci.com/paparazziuav/paparazzi/settings

flixr, 2018-02-04:
actually IMHO option 2 or 3 are preferable, as they will also work with the release tarballs we create

podhrmic, 2018-03-30:
FYI the Semaphore build now works (I did option 1). Not sure if we should keep this open in case somebody wants to do option 2 or 3.

gautierhattenberger, 2018-05-02:
@dewagter is it possible to try to replace `git am --signoff < ../fix_compiler_crash.patch` by `patch -p1 < ../fix_compiler_crash.patch` in the bebop_opencv makefile ?

closed 2018-05-04 by kirkscheper

podhrmic, 2018-05-04:
Thanks for fixing this @kirkscheper !

PR #2258 (2018-05-04): update opencv with multithread build
This should fix #2227 and reduce build time.