paparazzi/paparazzi #2228: Build dependencies (seems) not correctly setup

kirkscheper, 2018-02-05:
When running a make with -j8 on an eight core machine I get the following error:

[error.txt](https://github.com/paparazzi/paparazzi/files/1694479/error.txt)

It seems like that the dependencies of the pprzlink build is not correctly setup. I was looking into this specifically to try to speed up the travis build so that we can use as much of the build time for airframes. I took a quick look at it but not that experienced with makefiles so not too efficient at debugging. Can someone take a look at this and suggest a solution?

podhrmic, 2018-04-14:
I replicated the error. Anything beyond `make -j1` triggers the error. I am not sure what can be the problem, but it would be indeed great if we could build faster.

podhrmic, 2018-12-19:
The workaround doesn't work for me unfortunately:-(

PR #2472 (2019-12-11): [build] fix build dep for tools
fix #2228
commit 1b57ff13 (2019-12-11): [build] fix build dep for tools (#2472)

* [build] fix build dep for tools

fix #2228

* fix more dep for fast and recent proc

it's awesome

[image]
[image]
closed 2019-12-11 by OpenUAS