paparazzi/paparazzi #2231: find_vpaths.py: BOARD_CONFIG not defined after adding module

tomvand, 2018-02-07:
find_vpaths.py fails after adding the following module to examples/bebop.xml:
```
    <module name="cv_opencvdemo">
      <define name="OPENCVDEMO_CAMERA" value="front_camera"/>
    </module>
```
with the following error:
```
../../sw/tools/find_vpaths.py arm-linux-gnueabi-g++ /home/tom/paparazzi/var/aircrafts/bebop_orange_avoid/ap_tmp.list /home/tom/paparazzi
/home/tom/paparazzi/sw/airborne/arch/linux/mcu_periph/adc_arch.c:31:10: error: #include expects "FILENAME" or <FILENAME>
 #include BOARD_CONFIG
```

As far as I can figure out, this seems to be a problem with the var/aircrafts/.../nps_tmp.list file. After adding the cv_opencvdemo module, the required -DBOARD_CONFIG and other flags can no longer be found in this file.

I suspect the problem is in [Makefile.ac:173](https://github.com/paparazzi/paparazzi/blob/master/Makefile.ac#L173): the CFLAGS are not written to TMP_LIST. Adding the CFLAGS solves the missing defines problem, but causes a lot of warnings about C command line options being applied to C++ code.

Note that despite these errors, the compilation still seems to work.

gautierhattenberger, 2018-02-07:
I still don't like this find_vpath tool, and for sure it should take gcc and g++ as arguments and not only one for both types of files.

kevindehecker, 2018-02-28:
We all agree that nobody likes the find_vpaths hack. But actually, it is not the main problem but more of a symptom caused by these vpaths... So, it is a hack to fix another hack. If we can find a better solution to the vpaths or better yet get rid of those, or even more betterder if someone can explain what they are ment for in the first place... that would help!

gautierhattenberger, 2018-03-12:
It is just a way to not specify the global paths, so it "only" requires to fix all the makefiles to have a full path I guess.
Just one question: this is only needed for people use QTCreator right ?

kevindehecker, 2018-03-13:
Yes, only for QTC.

gautierhattenberger, 2018-03-13:
so at least with #2244 it is only an issue for people that actually need it

dewagter, 2018-03-24:
A work-around (for now) is to set PAPARAZZI_QT_GEN=

gautierhattenberger, 2021-12-13:
use proposed workaround if needed

closed 2021-12-13 by gautierhattenberger