paparazzi/paparazzi #2248: Support for Parrot Disco v1.4

AlessioDurante14, 2018-03-27:
Hello, 
I am a student of engineering at Politecnico di Milano working with your software.
Is there an update of Paparazzi which allows to work also with parrot disco at v1.4?
Thank you so much.

OpenUAS, 2018-03-27:
Yes. If I come around to it I'll add it to master. Tested with FW 1.4.1

AlessioDurante14, 2018-03-27:
Ok, thank you very much. 
Please when you are done, tell me what I should do to make the software work with v1.4.
I am really grateful to you

AlessioDurante14, 2018-03-27:
We have a campaign in which we need the drone just after the 1st week of April.
We would need the software for that time. Let me know if you think you are able to update it before that date. Thank you very much again.

OpenUAS, 2018-03-28:
Well I'll push en of today our PPRZ dev day it then you can tune it

AlessioDurante14, 2018-03-28:
Ok perfect! 
Please when you tune also the parameters, update the master with your results.
Thank you!

AlessioDurante14, 2018-04-04:
Hello, I'm sorry, I still can't find your update! Where did you push it? And how should I do to make it work?
Thanks!

OpenUAS, 2018-04-04:
Pushed to master, you can use e.g. ./start to choose the userconf/OPENUAS/openuas_conf.xml   or add vthe airframe and all its settinget to your own configuration XML

If I find some spare time will testfly n tune over weeken. Current could already fly if no ETECS is used. (Not yet set by default.) Weather permitting more tuning and frontcam fixes added (no promises ;)

closed 2018-04-04 by OpenUAS

AlessioDurante14, 2018-04-09:
Hello! 
I managed to connect Paparazzi to the disco.
I used the openuas_conf.xml and the tudelft control panel.
When I execute the flight UDP mode, the software says Disco, Wait GPS and I cannot go on with the flight planning.
How can I do? Thanks again

AlessioDurante14, 2018-04-09:
I can't connect the joystick either.
I'm using a Sony Playstation 3 joystick. 
Is it compatible? Which xml configuration should I use?