paparazzi/paparazzi #2251: Support for Pixhawk 4

podhrmic, 2018-04-14:
The release of Pixhawk 4 is planned for May: https://www.dronecode.org/pixhawk-4-coming-soon/

The question is, do we want Paparazzi to run on it? The hardware is similar to Chimera board (STM32F7 with an MPU unit), it has a bunch of IO, FBW co-processor and a SD card, so pretty much everything an autopilot needs.

The issues are:
- [ ] a couple of new sensors that need drivers
- [ ] likely only for ChibiOS, to utilize memory protection unit and SD card logging (so we would need to update the ChibiOS version first)
- [ ] porting ChibiOS to support a new board (pins, IO, clock setup etc)
- [ ] figure out the AP/FBW uploader (does it work for ChibiOS?)

Since I don't have any immediate need to use Pixhawk 4, I wanted to see if other are interested in using it?

noether, 2018-04-17:
I regularly meet people that are flying with PX4, and after discussing their mission goals, requirements, etc., they would be more than happy to try Paparazzi. In my opinion, they would do a couple of flight tests with Paparazzi if they do not need to change any hardware. It is evident that this support will expose Paparazzi to a broader community (maybe with more pros than cons).

We have a lot of PX4 installed on many vehicles in my institution, so we can test, provide feedback and develop specific code for PX4 eventually.

podhrmic, 2018-04-17:
Thanks for the input @noether 
Just to clarify - Paparazzi already runs on Pixhawk (at least rotorcraft code does), this is about supporting the new Pixhawk 4 (which is not available for purchase yet). So I guess the question is - do you think it is likely you would use the new Pixhawk 4 on your vehicles?

noether, 2018-04-17:
You are right. I was suspicious when I read STM32F7... I remember to try years ago Paparazzi in one PX4 (this four confused me :P) that I have on my desk years, and I was not able to make it work. Now I can see in the wiki than indeed, my PX4 is supported. So I will try it again.

As far as I know, we do not plan to buy this new PX4, since some people in the group bought loads of Pixhawk "cubes". So I do not think we will be able to help in the short term.

gautierhattenberger, 2018-04-17:
do we have more information about the sensors integrated to the board ?

podhrmic, 2018-04-17:
- 2x accel+gyro
- magnetometer
- baro

We have drivers for the baro, but the accel/gyro and mag are new sensors, all from different manufacturers (unfortunately I can't share the exact model of the sensors yet)

gautierhattenberger, 2018-04-18:
did they definitely drop open-source hardware or is it just for the design phase ?

OpenUAS, 2018-05-08:
For us a temp stop for hardware, not budget wise but just to much hardware and tasks, first need to finalize crazyflie 2, then still Parrot stuff and a new Parrot Drone https://www.parrot.com/eu/itscoming#itscoming etc. etc.

It is great that you look into it, although fixing more issue for Pixhawk IMHO better time spend, SD read Spectrum Softbind or time on e.g InterMCU from rotorcraft to repalce current fixedwing interMCu... well those d@mn 24 days ;)

podhrmic, 2018-05-22:
FYI I just got a beta test unit and it is actually a slick piece of hardware. Haven't done any testing besides looking at it, but it is slightly smaller and thinner than the old Pixhawk, and the connectors are super nice because they have a good latching mechanism.

OpenUAS, 2020-03-11:
Well I have 3 of them now, I might as well, assign the issue to me. Since at TU we fly several others with PPRZ already...

fvantienen, 2020-03-16:
This already works and I think most features are implemented

gautierhattenberger, 2020-03-16:
do you have everything in master or some PR should be made to complete the support ?

fvantienen, 2020-03-16:
Gonna look into that now and make some PRs

PR #2506 (2020-03-23): [hardware] Add Pixhawk 4 autopilot
This will close #2251 and needs #2505 

Not sure if everything is included in this pullrequest. Also no testing has been done with the PX4 flash modules, since I could not get it to work in vmware. Currently using the debug port with SWD to flash the autopilot and am not using the IO processor.
closed 2020-03-23 by gautierhattenberger