paparazzi/paparazzi #2252: [Airframe] "unit' with a define only sometiimes converts value

EwoudSmeur, 2018-04-17:
For things like the body to IMU rotation defines, it is custom to put the value in degrees (which makes sense from my point of view), and add unit="deg". This converts the value to radians, as the C code takes a value in radians.

Now I want to use a magnetometer to IMU rotation in the same way. I define for instance:
`<define name="HMC58XX_MAG_TO_IMU_PHI" value="6.5" unit="deg"/>`
But now the value is not converted to radians (nor do I get a warning that I am trying to use a rotation larger than 2pi)!

I think the behaviour of adding a unit should be uniform over the airframe file, as otherwise it is very confusing.

kirkscheper, 2018-04-17:
Where did you have that define? The unit conversions are not applied if it is in the firmware section. It is if you put it in the 'section' section.

This could be solved if the conversions are also applied in the firmware section @gautierhattenberger

EwoudSmeur, 2018-04-18:
You are right, they are in the firmware section...

OpenUAS, 2018-08-02:
This is indeed quite an issue if the plan is to move the modules to firmware section. Major label since it could crash you aircraft.

Cannot assign to self before there is some form of agreemen as a PPRZ team on how we want to handle it.

gautierhattenberger, 2018-12-05:
The defines in the firmware section are suppose to be compilation flags, so not really designed to handle the configurations of sensors or control loops. It is not even handled by the same generator.
One of the issue is that the DTD format is not rich enough to make the difference between these two types of defines. So it is not really possible to trigger an error at this level.
It is probably possible to include the code conversion in the firmware part, although it seems a bit tricky. Or eventually print a warning to tell that unit attrib will not the used.
Any thoughts ?

kirkscheper, 2018-12-05:
I would be fine with a warning as i am in favour of keeping the firmware section relatively empty.

closed 2018-12-06 by gautierhattenberger