paparazzi/paparazzi #2257: Support Parrot Disco FW 1.4

AlessioDurante14, 2018-05-03:
Hello,
We are the students involved in the photogrammetry project between Politecnico di Milano and Politecnico di Torino.
Sei have already asked for a support for the Parrot Disco FW 1.4.
We managed to connect to the disco with your release, but now, after pulling the master, the software doesn't work anymore with the disco 1.4.
Maybe some updates done in the recent days are in conflict with it.
Anyway, before this problem, the software didn't recognize the GPS signal, even if the disco received it. For this reason we could not go on with the flight planning...
Is there any option we forgot to set?
And then, how does the flight plan work? We launched open UAS but we didn't know how to use it (maybe because of the GPS). I cannot find any explanation for it.
Sorry for all these questions.
Hope you can solve our problems.
Thank you very much for your support.

Alessio Durante

AlessioDurante14, 2018-05-04:
Thank you for your help.
Anyway, in the issue #2248 I asked for support for the latest firmware version of the Parrot Disco, 1.4.1. 
With the version 5.12 you could only work with a Disco having the firmware up to 1.3, so I needed an update.
After my question in the issue #2248, a user called OpenUAS updated the master so that it could work also with the firmware 1.4.1. 
So I was using the master and it worked, except from the receiving of the GPS signal. 
Now if I keep using the master, it doesn't work anymore, in the sense that it doesn't even compile when I select 'Build' in the paparazzi center.

kirkscheper, 2018-05-04:
Can you give us a dump of the compiler errors? Just copy paste the result to a txt file and attach it here. We can't help if we don't know what the issue is exactly.

AlessioDurante14, 2018-05-04:
When I just push the button 'build' the compiler gives me the following messages:

CC /home/daniele/paparazzi/var/aircrafts/Disco/ap/modules/loggers/file_logger.o
modules/loggers/file_logger.c: In function 'file_logger_periodic':
modules/loggers/file_logger.c:98:29: error: 'COMMAND_THRUST' undeclared (first use in this function)
           stabilization_cmd[COMMAND_THRUST],
                             ^
modules/loggers/file_logger.c:98:29: note: each undeclared identifier is reported only once for each function it appears in
/home/daniele/paparazzi/conf/Makefile.linux:114: recipe for target '/home/daniele/paparazzi/var/aircrafts/Disco/ap/modules/loggers/file_logger.o' failed
make[1]: *** [/home/daniele/paparazzi/var/aircrafts/Disco/ap/modules/loggers/file_logger.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/daniele/paparazzi/sw/airborne'
Makefile.ac:282: recipe for target 'ap.compile' failed
make: *** [ap.compile] Error 2
make: Leaving directory '/home/daniele/paparazzi'

FAILED 'make -C /home/daniele/paparazzi -f Makefile.ac AIRCRAFT=Disco ap.compile' with code 2

I am using the master branch, with the OpenUAS configuration (OPENUAS/openuas_conf.xml) and the Disco settings. I am using then the ap compiler.
Thank you for your help.

gautierhattenberger, 2018-05-04:
you probably don't even need the `file_logger` module (which should only be used for rotorcraft at the moment) so you can remove it or comment it in your airframe file

AlessioDurante14, 2018-05-08:
[image]

Hello, 
I commented the rows you told me and it works.
Now the problem is that still the software cannot find the GPS of the drone.
I am sure the drone receives the GPS signal, it's the software that doesn't recognize it from the drone.
I have uploaded a screen-shot of what it is shown by Paparazzi.
Do you have ideas for the solution of this problem?

Another issue.
I can't even connect a joystick to paparazzi (maybe it can be the problem also for the GPS).
I am using a Play Station 3 joystick and I don't have an .xml configuration for it.
Can I run the autopilot without the joystick? 
Otherwise, where can I find a configuration for a Play Station 3 joystick?

Thank you again for your precious help.

Alessio

gautierhattenberger, 2018-05-09:
Do you have the `gps_ubx_ucenter` module loaded ? Unless this module have some bug, it is required to dynamical configure the GPS.

Regarding your joystick, you need to use this tool: http://wiki.paparazziuav.org/wiki/Input2Ivy
and in your airframe file, use the type `datalink` for the `radio_control` module. Please not that on the Disco, you can also use regular SBUS receiver directly connected to the Chuck autopilot, which is much more safer than doing RC through Wifi.

OpenUAS, 2018-05-31:
After last test Disco crashed. Have new one now and will retest after 20180531  but first will create a FWB solutuon with an additional LisaM onboard, since I2C mishaps are to be expected.

gautierhattenberger, 2018-06-06:
closed with 1d0b3c048b322c42eb7734129e55a68bd5362f9a

closed 2018-06-06 by gautierhattenberger