paparazzi/paparazzi #2267: Bug in 'Use throttle for arming'

matejkarasek, 2018-05-30:
The 'USE THROTTLE FOR MOTOR ARMING' only works the first time. Once throttle returns to zero, the 'autopilot.arming_status' gets stuck at 'AP_ARMING_STATUS_UNARMED_IN_AUTO'.

Introduced by this pull request: https://github.com/paparazzi/paparazzi/pull/2243 

@kevindehecker Can you look at it?

kevindehecker, 2018-05-30:
https://github.com/paparazzi/paparazzi/blob/cf9401e5dfcc7b668e9a2d40567b072e1923cc70/sw/airborne/firmwares/rotorcraft/autopilot_arming_throttle.h#L81

  } else if (autopilot_get_mode() != MODE_MANUAL) {
    autopilot.arming_status = AP_ARMING_STATUS_NOT_MODE_MANUAL;
  } else if (autopilot_unarmed_in_auto) {
    autopilot.arming_status = AP_ARMING_STATUS_UNARMED_IN_AUTO;

should probably become 

  } else if (autopilot_get_mode() != MODE_MANUAL || autopilot_unarmed_in_auto) {
    autopilot.arming_status = AP_ARMING_STATUS_NOT_MODE_MANUAL;

fvantienen, 2018-05-31:
Looking at the before #2243 that also does not seem to be correct. The following two lines are different: https://github.com/paparazzi/paparazzi/pull/2243/files#diff-8628a440d00db9fb98077e3cbcad9dedL111 and https://github.com/paparazzi/paparazzi/pull/2243/files#diff-8628a440d00db9fb98077e3cbcad9dedL102 (look at the `==` and `!=`).

I didn't look into detail, but at least it doesn't work as previous even with the fix from @kevindehecker.

gautierhattenberger, 2018-06-19:
@matejkarasek  @fvantienen do you have a tested pull request for this ?

OpenUAS, 2018-06-20:
@matejkarasek is on Holiday, will see if someone else can test it this week

PR #2283 (2018-06-27): fix throttle to arm
This should fix #2267. This fix has been tested.