paparazzi/paparazzi #2271: Cannot open video stream Bebop 2 even after downgrading to 3.3.0

bharat-robotics, 2018-06-08:
I am trying to save images and/or load video in Bebop 2, which originally had 4.4.0 firmware. I downgraded to 3.3.0 and used added rtp module as in [rtp_video_stream.xml](http://docs.paparazziuav.org/latest/module__video_rtp_stream.html). However, I cannot open the video. Am I missing something?

bharat-robotics, 2018-06-11:
I was able to capture images but the images are in incorrect format. I am getting images like this.
[image]

Has anyone encountered the problem?

OpenUAS, 2018-06-20:
At least firmware 3.4. But firmware 4.x is where we test for setting whitebalancee tc.. No one we've know has this problem. So best to ask on the mailing list ,but for now  wll lieave the issue open until we find time to validate.

@bharat-robotics , plz attatch you arframe file to this issue. There are setting for the camera, changed massivly beteen firmwares.

Things like:

    
    <define name="MT9F002_INITIAL_OFFSET_X" value="0.09" /> 
    <define name="MT9F002_INITIAL_OFFSET_Y" value="0." /> 
    <define name="MT9F002_OUTPUT_WIDTH" value="240" /> 
    <define name="MT9F002_OUTPUT_HEIGHT" value="520" /> 
    <define name="MT9F002_TARGET_FPS" value="30" />
    <define name="MT9F002_TARGET_EXPOSURE" value="20" />
    <define name="MT9F002_GAIN_GREEN1" value="4.0" />
    <define name="MT9F002_GAIN_GREEN2" value="4.0" />
    <define name="MT9F002_GAIN_RED" value="5.0" />
    <define name="MT9F002_GAIN_BLUE" value="5.0" />
    <define name="MT9F002_OUTPUT_SCALER" value="0.25"/>

OpenUAS, 2018-06-20:
BTW you basically have Two issues:

A) Cannot open video stream Bebop 2
B) Incorrect whitebalance or gain on frontcam

Q: @bharat-robotics did  you manage to solve A) ?

bharat-robotics, 2018-06-20:
I was able to solve the white balance by using the bebop_ae module. But not the video stream.

OpenUAS, 2018-06-21:
To fix your stream more info is needed, my advice is to ask on Gitter.im or in the mailing list. 
Also sharing how you fixed it here will help others, it a give and take ...

OpenUAS, 2018-07-24:
@bharat-robotics either a publik link to your repo, incl config you use, as in which airfame or add airframe to this issue.

OpenUAS, 2018-08-02:
@bharat-robotics Could not reproduce your issue. Feel free to reopen

closed 2018-08-02 by OpenUAS