paparazzi/paparazzi #2272: Drawn shapes on GCS blocks Waypoint edit

Dennis-Wijngaarden, 2018-06-19:
When drawing shapes on the GCS using the IVY SHAPE message, the shapes appear on the foreground of the GCS. It can hide the aircraft symbol as depicted in the picture. When waypoints are located underneath these shapes, I'm not able to edit those points from the GCS. 

In my opinion, the shapes should be plotted on the background of the GCS in order to overcome those problems.

[image]

podhrmic, 2018-06-19:
Agreed. @gautierhattenberger what do you think?

gautierhattenberger, 2018-06-20:
I agree shapes should be in a layer between maps and waypoints. I'll have a look.

gautierhattenberger, 2018-06-20:
@Dennis-Wijngaarden can you test #2274 ?

PR #2274 (2018-06-20): [gcs] add shapes to the background layer to keep WP and A/C on top
fix #2272
commit b8cdc091 (2018-06-20): [gcs] add shapes to the background layer to keep WP and A/C on top (#2274)

fix #2272
closed 2018-06-20 by OpenUAS